FreeOpcUa/opcua-modeler #2: XML Implementation Discussion

zerox1212, 2016-07-28:
Resume discussions on how to implement a UA model in XML from https://github.com/FreeOpcUa/python-opcua/issues/248 to keep issues between project separated.

oroulet, 2016-07-29:
> For example node.get_user_access_level() and node.get_parent() is missing. For now I'll use the latest master in my project.

We will need to add features for this project anyway, so working from master will be necessary.

oroulet, 2016-07-30:
inital UI done.

zerox1212, 2016-07-31:
Nice work.

When we export to XML, how can I make sure to only export the user nodes? I can't iterate over the address space otherwise the XML will have all the standard stuff. Is your UI keeping a list of everything that is created someplace?

oroulet, 2016-07-31:
That is the plan.... Need to modify some method calls in opcua lib to make
it happen... Will write a todo soon

zerox1212, 2016-08-15:
Now that we have some basic parts of the project  completed we should tackle problems on a per issue basis.

closed 2016-08-15 by zerox1212