FreeOpcUa/opcua-modeler #3: Master Plan

oroulet, 2016-07-31:
Work in progress, feel free to fix and/or comments @maljac @zerox1212 

- [x] initial UI taken from UA client
- [x] actions to add nodes
- [ ] Improve, add missing icons
- [ ] Implement namespace editor on top left (simle treeview showing NamespaceArray node with add and remove namespaces actions)
- [ ] import xml
- [ ] save xml (Do we need our own save/internal format in addition? )
- [ ] Use another font/color to clearly differentiate added nodes that will be exported
- [ ] keep track of all added nodes (need small modifications of python.opcua)
- [ ] finish UI shown when adding nodes, must be possible to specify node type and values for variables. Other attributes will be specified in Attribute editor
- [ ] finish Attribute editor, currently it only show attributes, apart from Value. Should use
- [ ] Finish reference editor
- [ ] Copy/past actions to copy nodes and attributes when mass creating nodes

zerox1212, 2016-08-01:
To avoid double work, I will take the 'save xml' task. I've already started it, but I don't have enough to start a PR yet. I'll try to get an initial PR to the opcua server this week.

oroulet, 2016-08-01:
@zerox1212 yes I let you have a look at that xml stuff. make a pr to python-opcua asap so other can comment early work.

And yes good idea to say who is working on what. I will polish the dialog to select a node and work on the attribute editor. Next is to correctly show nodes to export

zerox1212, 2016-08-03:
See this PR. https://github.com/FreeOpcUa/python-opcua/pull/270

zerox1212, 2016-08-15:
Does version 0.1 of the GUI support building methods?

oroulet, 2016-08-15:
Make it and it will :-)
No you should wait for 0.1, so that the basis works

But please write it in master plan so we do not forget it

oroulet, 2016-08-17:
@zerox1212 sounds crazy but it is now usable. Can you have a look and report?
you need to clone uamodeler and uawidgets repository and make a link

zerox1212, 2016-08-18:
I will grab the master and take a look. Thanks.

zerox1212, 2016-08-18:
Wow, it looks pretty good.

So far I see a few small issues. Some of the table columns can't be resized which is a bit annoying. Also, when you add a new variable, what does "auto data type" do?

oroulet, 2016-08-18:
Can you write the issues in master plan? No need to open bug request for
all small things for now. Yes the column resizing is an issue,  I do not
know how to fix it... If you know qt you can try.. Probably a settings of
the treeview

zerox1212, 2016-08-18:
I updated the plan. I did not know I could edit that.

zerox1212, 2016-08-18:
@maljac Can you test the modeler with some of the more complicated objects? I need help understanding what the XML should look like. At the moment XML export will probably not work on some configurations because of this.

zerox1212, 2016-09-29:
I have not looked at this in a while. What is the status of exporting methods? Turns out I now need the ability to build some simple methods in my model.

oroulet, 2016-10-01:
Method are exported correctly. But we should add a GUI to add automatically arguments

zerox1212, 2016-10-03:
When we change a custom type, such as object type, all instances of that type should update to reflect the changes in the type.

oroulet, 2016-10-03:
@zerox1212 I do not think so. expensive and very error prone. better let people write code

zerox1212, 2016-10-03:
UAModeler is not this way. If I have an instance of an object type in my model, and I add a new property to that object type, my instance automatically gets the new property.

If you have an address space with 500 instances of ObjectyType A, and you add a parameter to that type you will have to manually update all your instances. This would force you to do all your ObjectTypes first, and if you ever had to adjust it later it would be a major pain.

oroulet, 2016-10-04:
But the other way around if someone modify base object, should we change
the entire address space? That is going to be crazy expensive...

And if someone really has 500 instances, they really should make type in
xml then instanciate programmatically. Then won't get into any issue

jarkko-hautakorpi, 2019-08-06:
Is creating custom **Enumeration types (EnumStrings)** with values possible? UaModeler has EnumValues -section where you can add Acending/Mask/Gaps type of EnumValues.
Values are visible (like under ExceptionDeviationFormat) but not editable, even under own custom type.

oroulet, 2019-08-07:
You can just copy an existing one.