FreeOpcUa/opcua-modeler #4: [Feature] Generate relevant Python code for objects

maljac, 2016-08-10:
In UaModeler you can compile c / c++ code that are compliant with UA SDK. 
This could be a great feature in the opcua modeler too. Thus, you can create your information model and generate the necessary python objects that are needed if you want to use the model in the freeopcua-python context.

Example: Structure datatypes (see `BuildInfo` of standard spec as example)

oroulet, 2016-08-12:
@destogl was interested in that kind of things for events. That could be an idea. the prosys java sdk also does this,

zerox1212, 2016-08-18:
Do you have some ideas how to implement this?

We need some specification about how to make python classes from the UA model.

My opinion is that the python classes mirror the UA objects. I think we should create a decorator for every python class that mirrors a UA object. A decorated class would automatically subscribe to all children and include a local write method for python object > ua node.

destogl, 2016-09-01:
Hi guys. I am still interested in this kind of things. If I find a student to work on it next semester would be great :)

The idea from @zerox1212 is good.

But just for you to know: currently I have other stuff on my TODO list and unfortunately OPC UA is currently not in the middle of my work (that way I am not so active lately) but in month or so could be that I find someone to work on it...

oroulet, 2016-09-02:
@destogl sounds good, but as the uml export thing I think one should write that kind of things from the node structure in python-opcua, then add button in modeler. This is more flexible, it will support more use cases....