FreeOpcUa/opcua-modeler #5: [Feature] Generate UML from UA Model

maljac, 2016-09-02:
Hello, 

it would be great if we could generate some UML class diagramm from a UA model.

If you read a companion spec, you might have seen some UML diagramms that visualize the UA model. I think that these diagramms are a good way to get a fast understanding of the given model. Thus, if you have to write some documentation of your model, you would have a "single source of truth". One might even create some special extensions for the opcua-modeler nodes in order to document the model / type / ... nodes and create some sort of auto-documentation from the UA  model.

Since I am not familiar if there is some sort of XML specification for UML diagramms, I can not make an suggestion on technical details here.

oroulet, 2016-09-02:
Hei,Sounds like an interesting idea, but maybe we should generate that kind of things from the python node structure, that means in python-opcua, then just add a button in opcua-modeler.
@destogl that might be a good subject for a student?

destogl, 2016-09-07:
Yes! I will think about it. As I said it will be started first in a month or so