FreeOpcUa/opcua-modeler #6: Installation help

cpipero, 2016-09-19:
Love the effort, and would really like to test the tool.

I thought I installed everything needed but when I try running app.py it complains as follows:

`C:\Some\opcua-modeler-master>python app.py
Traceback (most recent call last):
  File "app.py", line 3, in <module>
    main()
  File "C:\Some\opcua-modeler-master\uamodeler\uamodeler.py", line 434, in main
    modeler = UaModeler()
  File "C:\Some\opcua-modeler-master\uamodeler\uamodeler.py", line 58, in __init__
    self._restore_state()
  File "C:\Some\opcua-modeler-master\uamodeler\uamodeler.py", line 182, in _restore_state
    self.restoreState(self.settings.value("main_window_state", b""))
TypeError: restoreState(self, Union[QByteArray, bytes, bytearray], version: int = 0): argument 1 has unexpected type 'sip.voidptr'`

of course, I have limited experience in python (but no one else is creating a free infomodel tool)...

oroulet, 2016-09-19:
Remove that line for the first start and it will work. I will check this
and make a new release soon

cpipero, 2016-09-19:
Thank you,

In case someone else runs into this, you have to comment out lines 181 through line 185 on the uamodeler.py file, basically almost all of the restore_state() function.

That works

oroulet, 2016-09-19:
Let me know how it goes. I am ready to fix bugs and improve things.

oroulet, 2016-09-19:
This is really a work in progress, but should work for models no using
extended objects

cpipero, 2016-09-19:
I can use the UI, but it doesn’t save anything I put it.

I wonder if there’s anything to do with a server opc.tcp://0.0.0.0:48400/freeopcua/uamodeler/ that seems to be required to run (maybe for model validation).
In the command prompt (under windows) it says “Listening on 0.0.0.0:48400” but when I try connecting a generic client I can’t find anything

<mailto:notifications@github.com%0b>>> <mailto:notifications@github.com>>

oroulet, 2016-09-19:
how did you install? You should probably install the master version of python-opcua, opcua-widgets and opcua-modeller by downloading from git.
I fixed a lot of things lately

oroulet, 2016-09-19:
pushed everything to pip. should be possible to install with pip install opcua-modeller if not bugs.....

cpipero, 2016-09-19:
I ran pip install but it told me that everything is up to date.
Still cannot save any modification.
I tried to create a simple ObjectType (with nothing in it) and saved it. No dice.

here's what the console output gives me:

`NewNodeBaseDialog returns: [2, 'SerialManagerType']
CURRENT C:/Users/Dromedaire/Downloads/opcua-modeler-master/OpenSCSNodeset.xml
Saving to C:/Users/Dromedaire/Downloads/opcua-modeler-master/OpenSCSNodeset.xml
Exporting  1 nodes: [Node(NumericNodeId(ns=2;i=1))]
and namespaces:  ['http://opcfoundation.org/UA/OpenSCS']
Exporting:  Node(NumericNodeId(ns=2;i=1))
NODE COMMON Node(NumericNodeId(ns=2;i=1))
Traceback (most recent call last):
  File "C:\Users\Dromedaire\AppData\Local\Programs\Python\Python35-32\lib\xml\etree\ElementTree.py", line 1078, in _escape_attrib
    if "&" in text:
TypeError: argument of type 'bool' is not iterable

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\Dromedaire\Downloads\opcua-modeler-master\uamodeler\uamodeler.py", line 296, in _save
    exp.dump_etree()  #debug
  File "C:\Users\Dromedaire\AppData\Local\Programs\Python\Python35-32\lib\site-packages\freeopcua-0.10.14-py3.5.egg\opcua\common\xmlexporter.py", line 79, in dump_etree
  File "C:\Users\Dromedaire\AppData\Local\Programs\Python\Python35-32\lib\xml\etree\ElementTree.py", line 1165, in dump
    elem.write(sys.stdout, encoding="unicode")
  File "C:\Users\Dromedaire\AppData\Local\Programs\Python\Python35-32\lib\xml\etree\ElementTree.py", line 775, in write
    short_empty_elements=short_empty_elements)
  File "C:\Users\Dromedaire\AppData\Local\Programs\Python\Python35-32\lib\xml\etree\ElementTree.py", line 940, in _serialize_xml
    short_empty_elements=short_empty_elements)
  File "C:\Users\Dromedaire\AppData\Local\Programs\Python\Python35-32\lib\xml\etree\ElementTree.py", line 932, in _serialize_xml
    v = _escape_attrib(v)
  File "C:\Users\Dromedaire\AppData\Local\Programs\Python\Python35-32\lib\xml\etree\ElementTree.py", line 1090, in _escape_attrib
    _raise_serialization_error(text)
  File "C:\Users\Dromedaire\AppData\Local\Programs\Python\Python35-32\lib\xml\etree\ElementTree.py", line 1056, in _raise_serialization_error
    "cannot serialize %r (type %s)" % (text, type(text).__name__)
TypeError: cannot serialize False (type bool)`

oroulet, 2016-09-20:
you found a bug :-( not very well tested yet.

oroulet, 2016-09-20:
The main issue is that I have no idea how to make tests with GUI applications. Do it is very easy to break something. I really need to find a solution for this.

oroulet, 2016-09-20:
fixed in master: https://github.com/FreeOpcUa/python-opcua/commits/master
This was a python-opcua exporter bug
Can you try to install from github then I do not need to make a new release every time I would like you to test something

cpipero, 2016-09-20:
Thank you it seems to be working now. I will keep testing and let you know
Really appreciated

oroulet, 2016-09-21:
you should update python-opcua library. fixed many things there today

closed 2016-09-21 by oroulet