FreeOpcUa/opcua-modeler #10: Load multiple XML but only export custom added nodes

Pro, 2016-09-20:
In OPC UA you can use multiple XML files as a basis and on top of them build your own information model.

E.g. I also want to use the "Devices" IM (https://opcfoundation.org/UA/schemas/DI/1.00/Opc.Ua.Di.NodeSet2.xml) for my modelling, and based on this build my own IM, let's say Namespace "urn:somestuff".
After modelling my own IM I only want to export the stuff I defined, not the whole "Devices" IM again, because it will be loaded separately into my OPC UA Server, so I only want to export the "urn:somestuff".

As far as I can see it that's currently not possible?

oroulet, 2016-09-20:
It is currently not, but as might have already noticed the code of opcua-modeller is very short, most of it is just ui stuff, so it is easy to add features.

Currently when importing an xml we flag them as new nodes. Internally we simply need to make the flagging optional.

The main issue is to decide on a good UI for defining "base" xml files. We can just put this in settings, but maybe you have a better proposition

there is also an issue with the "save" format. Currently we dot have our own format, we simply use the xml spec format. so we need to check if the "basis" reference xml can be saved in the xml format. I would like to avoid having to define our own "save" format

zerox1212, 2016-09-21:
It would be nice if you could help with features like this. I don't have enough experience with OPC UA to fully understand how to use multiple models.

oroulet, 2016-09-21:
thinking about it I thing we could have another widget on top of the namespace widget. that allows to import reference xml files.
Then we need a way to define referenced xml in xml format from opc foundation. Any suggestions @Pro ?

Pro, 2016-09-21:
I think the additional widget for referencing other base xml files would be the best option. This referenced xml file can then be handled the same way as you currently handle the main OPC UA xml (i.e., don't export it's nodes).

I think the easiest way to do this is if some of you adds this widget and I'll test it then. Just create a new feature branch and we can work there.

oroulet, 2016-09-21:
but when opening again a file create with a special referenced XML, a user would expect that the referenced file is also opened in the background, so one way or another we need a link to the referenced file in the generated xml... or use our own file format.... I would like to avoid the later.

btw  @Pro Do you know where default values of attributes like MinimumSamplingTime interval is defined? could not find it in spec...

Pro, 2016-09-21:
The only clean way is to create a simple project file format for the modeller (e.g. QSettings format) which contains these settings.
Adding custom data to the generated XML isn't allowed because then it doesn't conform to the linked .xsd file and thus the generated XML file isn't valid.

Hmm, isn't the MinimumSamplingTime dependend on the SDK? Just a guess...

oroulet, 2016-09-21:
MinimumSamplingTime seems to be specificified only sometimes.... and 0 and -1 are also possible values... which one is the default....

oroulet, 2016-09-22:
Added draft UI in master: https://github.com/FreeOpcUa/opcua-modeler/commit/8d80dad0e63e8a8ed3028c96f05fffe3db7db65c

Comments welcome
Not possible to save ref nodeset yet

Pro, 2016-09-22:
Can you set a minimum height for the widgets. Because I opened the gui and the TreeView was hidden. I had to move the splitter up. New user's won't find it.

oroulet, 2016-09-22:
it will be visible for new users, it was your saved eometry state that hidded it. But yes . minimum size is a good idea.

closed 2018-03-29 by oroulet