FreeOpcUa/opcua-modeler #13: Handle Methods Correctly

zerox1212, 2016-10-03:
I'm look at fixing methods.

When comparing to UAModeler, I see that the simple method I created requires an extension object.

Is this normal? Do we need to support extension objects to support methods?

Method Argument in UAModeler (296 = Argument):
*You can see that UAModeler only keeps track of what kind of extension object to make*
[image]

This is the XML that gets created.
[image]

zerox1212, 2016-10-04:
@oroulet As discussed in https://github.com/FreeOpcUa/opcua-widgets/issues/1; do you have time to modify xmlimporter.py to return created extension objects so that Attributes widget can display a list of ListOfExtensionObjects?

I can start working on xmlexporter.py to create an Argument extension object. It looks like a good place to start with extension objects.

oroulet, 2016-10-04:
> FreeOpcUa/opcua-widgets#1

Export and import of extension asking be fully implemented. But the UI is
not finished. I started but it is complex and I I had to focus on other
things ...

oroulet, 2016-10-04:
Should be fully.... Keyboard completion error

oroulet, 2016-10-04:
>>When comparing to UAModeler, I see that the simple method I created requires an extension object.

yes args are extension objects

zerox1212, 2016-10-04:
Should I try to do a basic UI? Perhaps just have 5 fixed fields for arguments instead of creating fields dynamically? Any field left blank is not used.

oroulet, 2016-10-04:
Probably a good idea to try. An arg has only 3 fields as far as I remember.
Name, description and type which is a node ID. Do not use ui files so it is
easier to extend later.

zerox1212, 2016-10-04:
I almost have it working. I will make a PR soon so you can take a look at it.

zerox1212, 2016-10-04:
So I'm having a hard time with a UaVariable which is an argument.

The problem is ```x.get_data_type()``` on this variable which has an extension object returns this:
[image]

Of course then the xml export doesn't work because the extension object isn't created. Any idea how to fix this? I've been trying for over and hour and I'm stuck.

oroulet, 2016-10-04:
What is with this data type?

zerox1212, 2016-10-04:
I'm finally to the point where I can work on xmlexporter, and so far it is not working. :)

```xml
  <UAVariable BrowseName="0:InputArguments" DataType="Argument" NodeId="i=20002" ParentNodeId="i=20001">
    <DisplayName>InputArguments</DisplayName>
    <Description>InputArguments</Description>
    <Value>
      <uax:ListOfArgument>
        <uax:Argument>Argument(Name:ArgName, DataType:TwoByteNodeId(i=NumericNodeId(i=10)), ValueRank:-1, ArrayDimensions:[], Description:LocalizedText(Encoding:2, Locale:None, Text:b'Desc'))</uax:Argument>
      </uax:ListOfArgument>
    </Value>
    <References>
      <Reference IsForward="false" ReferenceType="HasProperty">i=20001</Reference>
      <Reference ReferenceType="HasTypeDefinition">i=68</Reference>
    </References>
```

I will try to finish it up tomorrow.

zerox1212, 2016-10-05:
It's difficult to work in 3 separate repositories...

Methods are ready if you can finish UI.
Required by modeler to export method children:
https://github.com/FreeOpcUa/opcua-modeler/pull/14

Test widget to add a single input arg:
https://github.com/FreeOpcUa/opcua-widgets/pull/2

Updated exporter to handle extension objects:
https://github.com/FreeOpcUa/python-opcua/pull/311

zerox1212, 2016-10-07:
When a method is added, should the input and output argument nodes have a namespace?
[image]

oroulet, 2016-10-07:
Idx should be 0. This is correct

zerox1212, 2016-10-07:
Ok. New question.
If I make a node ns=1, shouldn't the browse name be "1:Name"? Currently browse name always gets "0:"
[image]

oroulet, 2016-10-07:
Browse name idx should be same as node ID by default

zerox1212, 2016-10-07:
Looks like it's a problem in manage_nodes.py. Instantiate is called, but it doesn't pass the namespace id.
```python
def create_object(parent, nodeid, bname, objecttype=None):
    """
    create a child node object
    arguments are nodeid, browsename, [objecttype]
    or namespace index, name, [objecttype]
    if objectype is given (a NodeId) then the type node is instantiated inclusive its child nodes
    """
    nodeid, qname = _parse_nodeid_qname(nodeid, bname)
    if objecttype is not None:
        objecttype = node.Node(parent.server, objecttype)
        return instantiate(parent, objecttype, nodeid, bname)[0]  # DOESN'T PASS NS ID?
```

zerox1212, 2016-10-07:
After more investigation. I think `instantiate(parent, objecttype, nodeid, bname)[0]` should pass `qname` not `bname`.

zerox1212, 2016-10-07:
>Idx should be 0. This is correct

I think this is not correct. Unifed Automation modeler does not do this, it adds the arguments to the same namespaces as the method.
[image]

The problem with how python modeler works, it will export the nodes, but if you import them again they get ignored.

oroulet, 2016-10-08:
As far as I remember, uaexpert refused to call the method if the arguments
where in another namespace than 0. But check and let us know what you find
out....

zerox1212, 2016-10-09:
This is very difficult to get working. If the arguments are not in NS=0 lots of wierd things happen. If I try to put them in NS=0 xmlparser doesn't work... I will keep trying to figure this out.

zerox1212, 2016-10-09:
From testing UA Expert has no problem calling methods with arguments in address space greater than 0. Now I just need to figure out what is going on with importing and calling them when XML is used...

zerox1212, 2016-10-10:
Many PRs but now everything seems to be working.

zerox1212, 2016-10-18:
Seems all related items have been merged. GUI needs some improvement, but it's functional. Closing...

closed 2016-10-18 by zerox1212