FreeOpcUa/opcua-modeler #17: Uri Export

zerox1212, 2016-10-07:
Why are Uris over 2 the only ones exported?
`uris = self.server.get_namespace_array()[2:]`

Shouldn't indexes of 1 or greater be exported?

oroulet, 2016-10-08:
We only export custom namespaces, not 0 and freeopcua s. But I remember writing code so freeopcua namespace is not created, and exporting everything over 1 not. 2…..... Did you read old code?

zerox1212, 2016-10-08:
I have latest code. It is still exporting NS 2; must be a bug.

zerox1212, 2016-10-09:
Should be fixed with https://github.com/FreeOpcUa/opcua-modeler/commit/d8ea8ea4124e575e9a4de5ef51dc55776b4b6a28

zerox1212, 2016-10-09:
This requires a fix in python-opcua. If my custom address space is in NS=1 it moves everything to NS=2 when importing for some reason?

zerox1212, 2016-10-09:
```python
elif name == 'NamespaceUris':
                for ns_index, ns_element in enumerate(child):
                    ns_uri = ns_element.text
                    ns_server_index = self.server.register_namespace(ns_uri)
                    self.namespaces[ns_index + 1] = (ns_server_index, ns_uri) # WHY ADD 1???
```
See comment, why do we always add 1?

oroulet, 2016-10-09:
That is because you have different URIs, if index point to another Uri then
we increase index. I do not know how robust code is, so there might be bugs

zerox1212, 2016-10-09:
I see now. The problem is that in python-opcua NS=1 is still consumed by "urn:freeopcua:python:server". I think the server should be NS=0?
[image]

oroulet, 2016-10-09:
I thought I fixed that. Look at commit log. We should not create freeopcua
namespace in modeler..

zerox1212, 2016-10-09:
It isn't  created in modeler, it's created in default server code.

server.py __init__():

```python
self.application_uri = "urn:freeopcua:python:server"

# setup some expected values
        self.register_namespace(self.application_uri)
        sa_node = self.get_node(ua.NodeId(ua.ObjectIds.Server_ServerArray))
        sa_node.set_value([self.application_uri])
```

So I guess it's actually my fault for not understanding how the namespaces work.

Thanks for the responses, sorry if I wasted your time.

closed 2016-10-09 by zerox1212

oroulet, 2016-10-09:
Yes but we remove it at startup in modeller..

zerox1212, 2016-10-09:
Modeler works, but then I was importing the XML from modeler into another server application that still had default code.