FreeOpcUa/opcua-modeler #18: Nodes with Children Don't Export

zerox1212, 2016-10-07:
Currently, adding nodes with children don't export all the nodes.

For example if you instantiate an Object from an Object type which has children, only the instantiated object is exported.

When an object is added, we need to search for children recursively in `_after_add` method.

zerox1212, 2016-10-07:
I will organize some PRs to fix this.

oroulet, 2016-10-08:
We could add an option to export _xml() to export nodes recursively...

What do you mean by instanciate? I thought I did not write code for
instantiation....

zerox1212, 2016-10-08:
If you choose a to create a new Object, and pick a custom data type, instantiation works by default. The only problem is the new children don't get added to the export list. I think I found a simple solution which i will make a PR for soon.

zerox1212, 2016-10-08:
https://github.com/FreeOpcUa/opcua-modeler/issues/18

With this creating a new object from a custom object type (that has multiple children) will now export the new object and all it's children correctly.

zerox1212, 2016-10-10:
Solved with your solution of calling instantiate directly.

closed 2016-10-10 by zerox1212