FreeOpcUa/opcua-modeler #24: support other internal servers (open62541 for example)

oroulet, 2016-11-13:
Should not be hard, I need to make a simple plugin architecture and use our client to access the internal server and optionaly support xml if server does not have xml support. It does require the server to support the node management service though!

@Pro Does open62541 has python wrapper? or should I use Cython? I coult not find any ref to python in your repository, but I kind of remember seing something..... I could also simply start the server as an external command but that will make it hard to kill it properly...

oroulet, 2016-11-13:
first attempt with ctypes lead to a coredump. Any suggestions?

     from ctypes import cdll
  
      if __name__ == "__main__":
           libua = cdll.LoadLibrary("./libopen62541.so")
          config = libua.UA_ServerConfig_standard
           nl = libua.UA_ServerNetworkLayerTCP(libua.UA_ConnectionConfig_standard, 16664)

coredump here

Pro, 2016-11-13:
open62541 only uses python for code generation from XML, there is no python wrapper as far as I know.
Starting the server externally shouldn't be a problem because you can use custom signals (e.g. SIGUSR1) and catch that in the open62541 implementation to shut it down properly.
I think that's the easiest solution.

I never worked with ctypes.
Also note that the current release RC1 has some problems with undefined stuff if you use the example code. I suggesst using the 0.2 branch.

And: your mention of jprf is wrong, the correct one would be jpfr ;)

oroulet, 2016-11-13:
> I think that's the easiest solution

Sounds boring ;-) we'll make styles worth

oroulet, 2016-11-13:
Styles -> ctypes .  stupid auto correction..

oroulet, 2016-11-13:
@Pro looks like I get a crash when calling things which are plugins... Are the plugins automatically loaded? Can i force loading?

Pro, 2016-11-13:
It depends how you build the library... There are various compile switches in CMake. I think you need to analyze the core dumps further to see where exactly it fails.

oroulet, 2016-11-17:
OK. ctypes works fine. It was a stupid typo on my side. but there are a lot of boring structs to expose, and possible fragility since structs are mined with so #ifdef that change structs at compile time.  I am considering looking at cython

jpfr, 2016-11-17:
I wrote a wrapper for open62541 and lua(jit).
That what quite a nice experience. It's like 2k lines of C- code to expose the OPC UA type system to native lua. However, that code will stay internal.

```lua
-- define a callback method
function hello_world(object_id, arg1)
    return "Hello " .. arg1.value
end

-- initialize the server with a port number
server = ua.Server(4842)

-- add a method node
string_arg = ua.types.Argument()
string_arg.dataType = ua.types.String.typeId
string_arg.valueRank = -1 -- argument is scalar
input_args = { string_arg } -- list of arguments
output_args = { string_arg }
attr = ua.types.MethodAttributes()
attr.displayName = ua.types.LocalizedText("", "Hello World")
attr.executable = true
browsename = ua.types.QualifiedName(1, "hello_world")
nodeid = ua.types.NodeId(1, "helloworld")
parent = ua.nodeIds.Objects
parent_reference = ua.nodeIds.HasComponent
server:addMethodNode(nodeid, parent, parent_reference,
    browsename, attr, hello_world,
    input_args, output_args)

server:run()
```

open62541 uses the generated data types quite a lot in the public API.
From my experience with the lua wrapper, it is better to write generic type handling functions that take the datatype description (the `UA_DataType`) as an additional input. For example by having python "objects" that contain a pointer to the C-data as well as the datatype struct.
Otherwise, there will be many thousands of lines of code just to get from Python to C-representations.

jpfr, 2016-11-17:
Overall, it might even be easier to write a smallish C-extension for Python.
https://docs.python.org/3/extending/building.html

oroulet, 2017-01-02:
Done using cython

closed 2017-01-02 by oroulet