FreeOpcUa/opcua-modeler #27: [0.3.1] Random crashes with module 'opcua.ua' has no attribute 'test_bit'

Stasik0, 2016-11-17:
Traceback (most recent call last):
  File "\WinPython-64bit-3.5.2.3Qt5\python-3.5.2.amd64
\lib\site-packages\uawidgets\attrs_widget.py", line 204, in createEditor
    return BitEditor(parent, attr, dv.Value.Value)
  File "\WinPython-64bit-3.5.2.3Qt5\python-3.5.2.amd64
\lib\site-packages\uawidgets\attrs_widget.py", line 30, in __init__
    if ua.test_bit(val, el.value):
AttributeError: module 'opcua.ua' has no attribute 'test_bit'

oroulet, 2016-11-17:
Thanks for the bug reports. This function has been moved...

oroulet, 2016-11-17:
this is fixed in master already

Stasik0, 2016-11-17:
Is there any way to pull master using pip?

oroulet, 2016-11-17:
no. you need to clone python-opcua, then opcua-modeller then opcua-widgets
as uawidget in opcua-modeller directory. Maybe I should write a script

Stasik0, 2016-11-17:
or just do a quick copy-paste FAQ, would be sufficient

zerox1212, 2016-11-17:
What I normally do is pull opcua-modeler, python-opcua, opcua-widgets to my local machine. Then I copy the opcua and widgets folder into the modeler folder. After that I run it from my IDE (run app.py).

Running it in debug will let you fix lots of these little bugs from this being an early release.

Stasik0, 2016-12-05:
@oroulet bump?

oroulet, 2016-12-05:
What is the status here? Have you managed to run master? This is supposed to be fixed there.
If you can test master and give feedback we can make a new release. We fixed many things

Stasik0, 2016-12-05:
Sorry, but is there some quick reference how to run master (i am familiar with git, but not python infrastructure)? It will probably be useful for other "bleeding edge"-tester as well.

zerox1212, 2016-12-05:
Python always searches for dependencies locally first, so if you just copy the `opcua` and `uawidgets` folder to the `opcua-modeler` folder python will use that, even if you have other libraries installed with pip.

oroulet, 2016-12-05:
you can do as @zerox1212 says or look at README file, I just updated it

Stasik0, 2016-12-08:
@oroulet I got the following error when running app.py

  File "\freeopcua-modeler\WinPython-64bit-3.5.2.3Qt5\
python-3.5.2.amd64\lib\site-packages\uawidgets\new_node_dialogs.py", line 6, in
<module>
    from opcua.common.ua_utils import dtype_to_vtype
ImportError: cannot import name 'dtype_to_vtype'

oroulet, 2016-12-08:
what code are you running, that method was remove a few weeks ago...
Try to run master by downloading with git as written in README

Stasik0, 2016-12-08:
i have downloaded master, I guess it takes the wrong dependency from site-packages and ignores pythonpath

oroulet, 2016-12-08:
probably a typo somewhere
try to write

    import uawidgets
    print(uawidhets.__file__) then you will see what module is imported

Stasik0, 2016-12-08:
yes, typo from my side, i got master running, will test it a bit

Stasik0, 2016-12-08:
please update readme with the line for windows
`set PYTHONPATH=%PYTHONPATH%;%cd%\python-opcua;%cd%\opcua-widgets`

oroulet, 2016-12-08:
Great!  It should have improved quite a lot since 0.3. I am waiting for
your feedback then we can release a new version

closed 2017-01-02 by oroulet