FreeOpcUa/opcua-modeler #28: [0.3.1] Unable to set value for attributes on type-level

Stasik0, 2016-11-17:
The indicator is showing type as "None" and crashes

showing error:  invalid literal for int() with base 10: 'None' 1
Traceback (most recent call last):
  File "\WinPython-64bit-3.5.2.3Qt5\python-3.5.2.amd64
\lib\site-packages\uawidgets\attrs_widget.py", line 237, in setModelData
    dv.Value = string_to_variant(text, dtype)
  File "C:\Users\DESTGRU\Downloads\WinPython-64bit-3.5.2.3Qt5\python-3.5.2.amd64
\lib\site-packages\opcua\common\ua_utils.py", line 113, in string_to_variant
    return ua.Variant(string_to_val(string, vtype), vtype)
  File "\WinPython-64bit-3.5.2.3Qt5\python-3.5.2.amd64
\lib\site-packages\opcua\common\ua_utils.py", line 84, in string_to_val
    val = int(string)
ValueError: invalid literal for int() with base 10: 'None'

I have to add <Value> Node in XML manually

oroulet, 2016-11-17:
can you explain again ?

oroulet, 2016-11-18:
you mean that you create a node variable but somehow manage to set its value to Null when creating it? How do you do that? This should only be possible when creating a extention object, and we create them yet anyway (we can edit them)

Stasik0, 2016-11-18:
Maybe by copy-paste?

oroulet, 2016-11-18:
it is possible to change type of a variable node by first modyfing its datatype then its value, which will then use the new datatype...maybe we have a bug here we should always use the datatype and not the type of the variant.... @zerox1212

Stasik0, 2016-11-18:
I ran into it by copy-pasting attributes from NS0, i think

zerox1212, 2016-11-18:
Can you give me a step by step of what you did? I never used variable types before, and I generally didn't use copy paste.

I can tell you that when you create a new variable, it uses the data type from the selection to set the node data type and the code uses the same data type to create a variant of that type. Whatever is typed in the value box is assumed to match what the user selects, I don't think there is any protection here.

@oroulet , it seems like he is saying that copy and paste isn't saving the value/dtype properly. That is the only way dtype could end up as None (which is different than ua.Null). Perhaps we should check for this condition, and set the node datatype to ua.Null, then the code would create a Null variant as well.

Stasik0, 2016-11-21:
You get a "None" type when importing XML nodesets.  Open Types->ObjectTypes->FileType->MimeType->Value. Now i see no possibility to set some other type than "None"

oroulet, 2016-12-05:
This is fixed as far as I remember try master and close issue.