FreeOpcUa/opcua-modeler #31: [0.3.1] Copy-Paste Node does not modify Namespace id of browename and nodeid

Stasik0, 2016-11-17:
Copy-paste should be of paste-environment and adjust Browsename acoringly

oroulet, 2016-11-17:
Can you explain further you explanantion? when copying a new nodeid is
created and the original browse name is kept. how do you think it should
behave?

Stasik0, 2016-11-17:
There is a Browsename in format "nsid:String" and there is a nodeid as a struct with nsid and so on.

If I paste a node where a context is clear e.g. an attribute form node in NS0 and paste it as a child of type in NSX, I'd like that browsename is changed from "0:bla" to "X:bla" and the NS id in the new nodeid is change accordingly. Furthermore, a popup would be good to ask me to enter or generate a unique nodid in new namespace.

What do you think?

oroulet, 2016-11-17:
Showing a dialog is a good idea. We can ask for namespace. But asking for
node IDs seems difficult, there may be many nodes created...

zerox1212, 2016-11-17:
To be honest I never use copy paste when I'm using modeler, but it seems like we would only need to choose a namespace for pasting. Then we can use the existing "new node" convention, which is like this:
1. Auto node id uses counter specific for each address space
2. Create own node id by typeing, and you need to type the "ns=x" yourself; if you make an int or string node id will affect child node ids

In a previous PR I tired to make ns=x automatically match the selected namespace, but @oroulet said that user should type it. I don't really understand why, but that's how it is for now.