FreeOpcUa/opcua-modeler #34: Add Unit Test for GUI

zerox1212, 2016-11-18:
If I get some free time in the next weeks I was thinking about expanding my knowledge on unit tests. I have never wrote them before.

I read this and it doesn't look that difficult, but I think we should add this to modeler.

Here is some info, but I will probably need help because I'm inexperienced.
http://johnnado.com/pyqt-qtest-example/

oroulet, 2016-11-18:
The approach described in this page of fundamentally broken. Events are not sent. So you cannot test your application.
I think we need to look at solutions like frologic. Can you find one that is free for open source?

zerox1212, 2016-11-18:
Seems like there are not many options.

Some people used this: http://www.sikuli.org/

I also found this, which looks interesting, but not sure if it will work: http://pyqt.sourceforge.net/Docs/dip/automate_tutorial.html

For now I guess we could skip testing the actual GUI and just write some unit tests for the functions themselves with typical user data...

oroulet, 2016-11-19:
Could not find anything better. I just added a test using the approach of http://johnnado.com/pyqt-qtest-example/
You can have a look. We ca do a bunch of testing that way, but we cannot test everything...

oroulet, 2016-11-19:
need to integrate it to travis and github...

oroulet, 2016-11-19:
Now I managed to integrate it. you can write tests!

oroulet, 2016-12-10:
kind of works now

closed 2016-12-10 by oroulet