FreeOpcUa/opcua-modeler #36: [master] Adding namespace results in a crash

Stasik0, 2016-12-09:
I run master. New project -> right-click "Add namespace" -> "test" as value

```
Traceback (most recent call last):
  File "C:\Users\z\Documents\freeopcua-modeler\opcua-widgets\uawidgets\uti
ls.py", line 21, in wrapper
    result = func(self, *args)
  File "C:\Users\z\Documents\freeopcua-modeler\opcua-modeler\uamodeler\nam
espace_widget.py", line 127, in setModelData
    self.logger.info("Writting namespace array: %s", uries)
AttributeError: 'MyDelegate' object has no attribute 'logger'

During handling of the above exception, another exception occurred:

TypeError: NamespaceWidget.error[str].emit(): argument 1 has unexpected type 'At
tributeError'
```

oroulet, 2016-12-09:
fixed stupid loggin error and added a test, so it does not happen again

oroulet, 2016-12-09:
git pull in opcua-modeler to uddate

closed 2016-12-09 by oroulet