FreeOpcUa/opcua-modeler #39: [0.4] Saving Symmetric Reference Types does not work

Stasik0, 2017-04-05:
Hi, i got a symmetric reference type without inverseName

```
  <UAReferenceType BrowseName="1:zz" NodeId="ns=1;i=4002" Symmetric="true">
    <DisplayName>zz</DisplayName>
    <References>
      <Reference IsForward="false" ReferenceType="HasSubtype">i=32</Reference>
    </References>
  </UAReferenceType>
```
Saving gives an error:
  File "C:\freeopcua-modeler\python-opcua\opcua\common\xmlexporter.py", line 299, in add_etree_reference_type
    self._add_sub_el(obj_el, 'InverseName', var.Value.Value.Text.decode('utf-8'))
AttributeError: 'NoneType' object has no attribute 'decode'

oroulet, 2017-04-24:
https://github.com/FreeOpcUa/python-opcua/pull/438 should fix it. Let me know is something else is missing. I do not really understand the inversename stuff

oroulet, 2017-04-26:
reopen if issue still exists

closed 2017-04-26 by oroulet