FreeOpcUa/opcua-modeler #41: [0.4] How to load additional NodeSets

Stasik0, 2017-04-05:
I have a dedicated files for instance (i.xml) and type node (t.xml) sets. How can I load t.xml first and i.xml later and work in it? Currently I can add both of them as reference node sets, but then opening a new file reloads the whole server.

oroulet, 2017-04-06:
Can you explain a bit more what you are trying to achieve?

Stasik0, 2017-04-06:
Okay, I have a companion standard describing types (t.xml) and an own node set with instances of types described in (i.xml). Now, I cannot load and edit i.xml directly since the types are not there. Adding "t.xml" as a "reference node set" helps, however the existing instance in i.xml are not show since they were loaded before including t.xml

oroulet, 2017-04-06:
OK, we have an issue here. Does your i.xml file has a reference to t.xml somewhere? We could not find it in spec, but maybe that information is under <model> tag? 

maybe as a bypass you can import i.xml after adding reference? I do not remember if the functionnality was exposed.

Stasik0, 2017-04-06:
it has an entry as <NamespaceUris>

I can of course import i.xml after reference, but then I am editing and exporting another file (that i currently merge in i.xml manually)

oroulet, 2017-06-11:
I really wanted to avoid having to create our own file format but it looks like we have too to solve this...

zerox1212, 2017-06-12:
Why do you need two XML files when you can just put your types in a separate name space?

oroulet, 2017-06-20:
added to master plan https://github.com/FreeOpcUa/opcua-modeler/issues/3 This is critical

Stasik0, 2017-06-21:
Types are typically goverend by someone else (companion standards). I dont really want to mix classes and instances in one file.

oroulet, 2017-07-03:
fixed in https://github.com/FreeOpcUa/opcua-modeler/pull/51
Need more testing before merge as I add to add a new file type '.uamodel'

oroulet, 2018-03-29:
This now merged and seems to work

closed 2018-03-29 by oroulet