FreeOpcUa/opcua-modeler #42: Open XML functionality does not work

alin256, 2017-04-05:
I have installed a version with pip3.
I expect that this version should be able to open example.xml.
However I get the following error: 

uamodeler.uamodeler - INFO - Starting server on opc.tcp://0.0.0.0:48400/freeopcua/uamodeler/')
opcua.server.address_space - WARNING - AddNodesItem: Requested NodeId NumericNodeId(i=2001) already exists')
uawidgets.utils - ERROR - The requested node id is already used by another node.(BadNodeIdExists)')
Traceback (most recent call last):
  File "/usr/local/lib/python3.4/dist-packages/uawidgets/utils.py", line 21, in wrapper
    result = func(self, *args)
  File "/usr/local/lib/python3.4/dist-packages/uamodeler/uamodeler.py", line 357, in _open
    path = self._import(path)
  File "/usr/local/lib/python3.4/dist-packages/uamodeler/uamodeler.py", line 334, in _import
    new_nodes = self.server_mgr.import_xml(path)
  File "/usr/local/lib/python3.4/dist-packages/uamodeler/server_manager.py", line 72, in import_xml
    return self._backend.import_xml(path)
  File "/usr/local/lib/python3.4/dist-packages/uamodeler/server_manager.py", line 105, in import_xml
    return self._server.import_xml(path)
  File "/usr/local/lib/python3.4/dist-packages/opcua/server/server.py", line 422, in import_xml
    return importer.import_xml(path)
  File "/usr/local/lib/python3.4/dist-packages/opcua/common/xmlimporter.py", line 63, in import_xml
    node = self.add_object_type(nodedata)
  File "/usr/local/lib/python3.4/dist-packages/opcua/common/xmlimporter.py", line 171, in add_object_type
    res[0].StatusCode.check()
  File "/usr/local/lib/python3.4/dist-packages/opcua/ua/uatypes.py", line 201, in check
    raise UaStatusCodeError(self.value)
opcua.ua.uaerrors._auto.BadNodeIdExists: The requested node id is already used by another node.(BadNodeIdExists)
uamodeler.uamodeler - INFO - Starting server on opc.tcp://0.0.0.0:48400/freeopcua/uamodeler/')
opcua.server.address_space - WARNING - AddNodesItem: Requested NodeId NumericNodeId(i=2001) already exists')
uawidgets.utils - ERROR - The requested node id is already used by another node.(BadNodeIdExists)')
Traceback (most recent call last):
  File "/usr/local/lib/python3.4/dist-packages/uawidgets/utils.py", line 21, in wrapper
    result = func(self, *args)
  File "/usr/local/lib/python3.4/dist-packages/uamodeler/uamodeler.py", line 357, in _open
    path = self._import(path)
  File "/usr/local/lib/python3.4/dist-packages/uamodeler/uamodeler.py", line 334, in _import
    new_nodes = self.server_mgr.import_xml(path)
  File "/usr/local/lib/python3.4/dist-packages/uamodeler/server_manager.py", line 72, in import_xml
    return self._backend.import_xml(path)
  File "/usr/local/lib/python3.4/dist-packages/uamodeler/server_manager.py", line 105, in import_xml
    return self._server.import_xml(path)
  File "/usr/local/lib/python3.4/dist-packages/opcua/server/server.py", line 422, in import_xml
    return importer.import_xml(path)
  File "/usr/local/lib/python3.4/dist-packages/opcua/common/xmlimporter.py", line 63, in import_xml
    node = self.add_object_type(nodedata)
  File "/usr/local/lib/python3.4/dist-packages/opcua/common/xmlimporter.py", line 171, in add_object_type
    res[0].StatusCode.check()
  File "/usr/local/lib/python3.4/dist-packages/opcua/ua/uatypes.py", line 201, in check
    raise UaStatusCodeError(self.value)
opcua.ua.uaerrors._auto.BadNodeIdExists: The requested node id is already used by another node.(BadNodeIdExists)
uamodeler.uamodeler - INFO - Starting server on opc.tcp://0.0.0.0:48400/freeopcua/uamodeler/')
uamodeler.uamodeler - INFO - Starting server on opc.tcp://0.0.0.0:48400/freeopcua/uamodeler/')
opcua.server.address_space - WARNING - AddNodesItem: Requested NodeId NumericNodeId(i=2001) already exists')
uawidgets.utils - ERROR - The requested node id is already used by another node.(BadNodeIdExists)')
Traceback (most recent call last):
  File "/usr/local/lib/python3.4/dist-packages/uawidgets/utils.py", line 21, in wrapper
    result = func(self, *args)
  File "/usr/local/lib/python3.4/dist-packages/uamodeler/uamodeler.py", line 357, in _open
    path = self._import(path)
  File "/usr/local/lib/python3.4/dist-packages/uamodeler/uamodeler.py", line 334, in _import
    new_nodes = self.server_mgr.import_xml(path)
  File "/usr/local/lib/python3.4/dist-packages/uamodeler/server_manager.py", line 72, in import_xml
    return self._backend.import_xml(path)
  File "/usr/local/lib/python3.4/dist-packages/uamodeler/server_manager.py", line 105, in import_xml
    return self._server.import_xml(path)
  File "/usr/local/lib/python3.4/dist-packages/opcua/server/server.py", line 422, in import_xml
    return importer.import_xml(path)
  File "/usr/local/lib/python3.4/dist-packages/opcua/common/xmlimporter.py", line 63, in import_xml
    node = self.add_object_type(nodedata)
  File "/usr/local/lib/python3.4/dist-packages/opcua/common/xmlimporter.py", line 171, in add_object_type
    res[0].StatusCode.check()
  File "/usr/local/lib/python3.4/dist-packages/opcua/ua/uatypes.py", line 201, in check
    raise UaStatusCodeError(self.value)
opcua.ua.uaerrors._auto.BadNodeIdExists: The requested node id is already used by another node.(BadNodeIdExists)
uamodeler.uamodeler - INFO - Starting server on opc.tcp://0.0.0.0:48400/freeopcua/uamodeler/')
uamodeler.uamodeler - INFO - Starting server on opc.tcp://0.0.0.0:48400/freeopcua/uamodeler/')
opcua.server.address_space - WARNING - AddNodesItem: Requested NodeId NumericNodeId(i=2001) already exists')
uawidgets.utils - ERROR - The requested node id is already used by another node.(BadNodeIdExists)')
Traceback (most recent call last):
  File "/usr/local/lib/python3.4/dist-packages/uawidgets/utils.py", line 21, in wrapper
    result = func(self, *args)
  File "/usr/local/lib/python3.4/dist-packages/uamodeler/uamodeler.py", line 357, in _open
    path = self._import(path)
  File "/usr/local/lib/python3.4/dist-packages/uamodeler/uamodeler.py", line 334, in _import
    new_nodes = self.server_mgr.import_xml(path)
  File "/usr/local/lib/python3.4/dist-packages/uamodeler/server_manager.py", line 72, in import_xml
    return self._backend.import_xml(path)
  File "/usr/local/lib/python3.4/dist-packages/uamodeler/server_manager.py", line 105, in import_xml
    return self._server.import_xml(path)
  File "/usr/local/lib/python3.4/dist-packages/opcua/server/server.py", line 422, in import_xml
    return importer.import_xml(path)
  File "/usr/local/lib/python3.4/dist-packages/opcua/common/xmlimporter.py", line 63, in import_xml
    node = self.add_object_type(nodedata)
  File "/usr/local/lib/python3.4/dist-packages/opcua/common/xmlimporter.py", line 171, in add_object_type
    res[0].StatusCode.check()
  File "/usr/local/lib/python3.4/dist-packages/opcua/ua/uatypes.py", line 201, in check
    raise UaStatusCodeError(self.value)
opcua.ua.uaerrors._auto.BadNodeIdExists: The requested node id is already used by another node.(BadNodeIdExists)

zerox1212, 2017-04-05:
I tested this and it does have a problem. I'm not sure why it thinks there is an existing node ID though. I have to debug it.

oroulet, 2017-04-05:
What example. Xml? You cannot import a file reusing a nodeid that exists in
standard address space

zerox1212, 2017-04-05:
He is trying to open this xml that comes with modeler: https://github.com/FreeOpcUa/opcua-modeler/blob/master/example.xml

I remade it and now it works, I don't know why the original example complains, I did not see any duplicated IDs, could be a namespace issue.

```XML
<?xml version='1.0' encoding='utf-8'?>
<UANodeSet xmlns="http://opcfoundation.org/UA/2011/03/UANodeSet.xsd" xmlns:uax="http://opcfoundation.org/UA/2008/02/Types.xsd" xmlns:xsd="http://www.w3.org/2001/XMLSchema" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance">
  <Aliases>
    <Alias Alias="HasTypeDefinition">i=40</Alias>
    <Alias Alias="HasSubtype">i=45</Alias>
    <Alias Alias="HasProperty">i=46</Alias>
    <Alias Alias="HasComponent">i=47</Alias>
    <Alias Alias="Organizes">i=35</Alias>
    <Alias Alias="Float">i=10</Alias>
    <Alias Alias="String">i=12</Alias>
  </Aliases>
  <NamespaceUris>
    <Uri>urx:mycompany:myproject</Uri>
  </NamespaceUris>
  <UAObjectType BrowseName="1:MyDevice" NodeId="ns=1;i=2001">
    <DisplayName>MyDevice</DisplayName>
    <Description>MyDevice</Description>
    <References>
      <Reference IsForward="false" ReferenceType="HasSubtype">i=58</Reference>
      <Reference ReferenceType="HasComponent">ns=1;i=2005</Reference>
    </References>
  </UAObjectType>
  <UAObject BrowseName="1:obj1" NodeId="ns=1;i=2002" ParentNodeId="i=85">
    <DisplayName>BaseObjectType</DisplayName>
    <Description>The base type for all object nodes.</Description>
    <References>
      <Reference IsForward="false" ReferenceType="Organizes">i=85</Reference>
      <Reference ReferenceType="HasTypeDefinition">i=58</Reference>
      <Reference ReferenceType="HasComponent">ns=1;i=2003</Reference>
      <Reference ReferenceType="HasProperty">ns=1;i=2004</Reference>
      <Reference ReferenceType="HasComponent">ns=1;i=2006</Reference>
    </References>
  </UAObject>
  <UAVariable BrowseName="1:var1" DataType="Float" NodeId="ns=1;i=2003" ParentNodeId="ns=1;i=2002">
    <DisplayName>var1</DisplayName>
    <Description>var1</Description>
    <Value>
      <uax:Float>9.99</uax:Float>
    </Value>
    <References>
      <Reference IsForward="false" ReferenceType="HasComponent">ns=1;i=2002</Reference>
      <Reference ReferenceType="HasTypeDefinition">i=63</Reference>
    </References>
  </UAVariable>
  <UAVariable BrowseName="1:myproperty" DataType="String" NodeId="ns=1;i=2004" ParentNodeId="ns=1;i=2002">
    <DisplayName>myproperty</DisplayName>
    <Description>myproperty</Description>
    <Value>
      <uax:String>property_value</uax:String>
    </Value>
    <References>
      <Reference IsForward="false" ReferenceType="HasProperty">ns=1;i=2002</Reference>
      <Reference ReferenceType="HasTypeDefinition">i=68</Reference>
    </References>
  </UAVariable>
  <UAVariable BrowseName="1:DeviceName" DataType="String" NodeId="ns=1;i=2005" ParentNodeId="ns=1;i=2001">
    <DisplayName>DeviceName</DisplayName>
    <Description>DeviceName</Description>
    <Value>
      <uax:String>MyName</uax:String>
    </Value>
    <References>
      <Reference IsForward="false" ReferenceType="HasComponent">ns=1;i=2001</Reference>
      <Reference ReferenceType="HasTypeDefinition">i=63</Reference>
    </References>
  </UAVariable>
  <UAVariable ArrayDimensions="3" BrowseName="1:varList" DataType="String" NodeId="ns=1;i=2006" ParentNodeId="ns=1;i=2002">
    <DisplayName>varList</DisplayName>
    <Description>varList</Description>
    <Value>
      <uax:ListOfString>
        <uax:String>one</uax:String>
        <uax:String>two</uax:String>
        <uax:String>three</uax:String>
      </uax:ListOfString>
    </Value>
    <References>
      <Reference IsForward="false" ReferenceType="HasComponent">ns=1;i=2002</Reference>
      <Reference ReferenceType="HasTypeDefinition">i=63</Reference>
    </References>
  </UAVariable>
</UANodeSet>
```

oroulet, 2017-04-06:
Commit it. This file was outdated anyway

oroulet, 2017-04-06:
can you commit an updated example? it was completely outdated anyway.

zerox1212, 2017-04-13:
We should probably make a few more examples, especially using node sets. However I have never used node sets so would be great if someone could help.

closed 2017-04-26 by oroulet