FreeOpcUa/opcua-modeler #46: Could it cread ModellingRule?

yealfor, 2017-05-16:
I created my address space, but there is no ModellingRule in the xml. When I  implement the  new model with open62541(a opc ua sdk), it can't present the correct variant , because that variant has no ModellingRule. So help！

zerox1212, 2017-05-16:
I also do not know how to use ModellingRule. If you can explain we can probably add it.

yealfor, 2017-05-17:
This is a variant which can be implemented in address space:
<UAVariable BrowseName="1:State" DataType="Float" NodeId="ns=1;i=2008" ParentNodeId="ns=1;i=2005">
		<DisplayName>State</DisplayName>
		<Description>State</Description>
		<References>
			<Reference IsForward="false" ReferenceType="HasComponent">ns=1;i=2005</Reference>
			<Reference ReferenceType="HasModellingRule">i=78</Reference>
			<Reference ReferenceType="HasTypeDefinition">i=63</Reference>
		</References>
		<Value>
			<uax:Float>9.99</uax:Float>
		</Value>
	</UAVariable>
where the reference(<Reference ReferenceType="HasModellingRule">i=78</Reference>)  can't be added in free uamodeler. 
As I know, the ModellingRule includes Mandatory and Optional at least. If the ModellingRule of a variant is not Mandatory, the address space may not implement it, but there is a need to implement that variant.

oroulet, 2017-05-17:
OK, HasModellingRule is a type reference and reference editor is not implemented yet. But this is very high on my todo list. Should have a look..

zerox1212, 2017-05-18:
Do you have anything started with this? If you have some code already I might have time to help out with this.

oroulet, 2017-05-19:
I have a lot of copy paste code. I need to make the basic work, then I can publish for testing/improvement

oroulet, 2017-05-19:
started work here. https://github.com/FreeOpcUa/opcua-widgets/pull/8  
not finished. I am not sure how to modify a reference. Looks like I need to delete and rewrite it

zerox1212, 2017-05-19:
I'll take a look at it.

oroulet, 2017-05-20:
ready for testing now. I have not tested much myself, but it should work. suggestions to improve and PR are welcome

oroulet, 2017-05-20:
and merged to master. So now it is easy to test

closed 2017-06-11 by oroulet