FreeOpcUa/opcua-modeler #50: First namespace

Darjohka, 2017-06-30:
I have a problem reading a value from the internal OPC server. I use the QuickOPC toolkit and get the error message:

`"The first namespace in the table must be the OPC-UA namespace."`

After loading example.xml the namespace array contains:
    `[0] urn:freeopcua:python:server`
    `[1] urx:mycompany:myproject`

[image]

The screenshot that comes with the folder opcua-modeler (screenshot.png) shows index 0 with the value `http://opcfoundation.org/UA/`.

So, is the _index 0 problem_ a bug or can I configure it somehow?

Since I used `git clone`, I assume,  I have the most recent version ...

Best regards!

zerox1212, 2017-06-30:
I think it's a bug. If you open modeler and just start with a brand new model, then add some of your stuff and export it does it work?

Darjohka, 2017-06-30:
Starting with a brand new model leads to the same problem ...

oroulet, 2017-07-01:
When do you get that error? It looks like we remove the address space 0 in modeller. I do not remember why... there was a reason.... Anyway when you export xml, it should not change anything since when importing it, the server will allready have namespace 0 defined...

oroulet, 2017-07-01:
there was a bug, fixed now in master

Darjohka, 2017-07-01:
Sounds great! I'm away on business. Next Thursday I will have time to test the connection with QuickOPC toolkit ... Thank you in advance!

Darjohka, 2017-07-10:
Problem solved! Thank you very much!

closed 2017-07-10 by Darjohka