FreeOpcUa/opcua-modeler #54: Not able to create a property under a variable

w4tsn, 2018-03-29:
Is this a design decision or expected behaviour? I was expecting to be able to create PropertyType-Variables under other Variables. In the documentation I've only read that PropertyType-Variables cannot be placed under other PropertyType-Variables. So does this apply to all variables?

oroulet, 2018-03-29:
Very good question. We disabled creation of some object types at some places but it is probably wrong at some places. Can you explain s bit more what you want to do? Maybe screenshot? Or/and ref to spec?

w4tsn, 2018-03-29:
I want to create a structure where the Variables of an Object are complex, which means they themselves have Variables and Properties.

I now found evidence [in this whitepaper (on page 10)](http://www.ascolab.com/images/stories/ascolab/doc/ua_whitepaper_technicaloverview_e.pdf) that it should be possible to create variables and properties under other variables (excerpt):

|Type Definitions|Comment|
|-|-|
|VariableTypes|Data variables (typed) represent data of an object, can be complex (have other data variables) and have properties. Properties (identified by BrowseName) represent characteristics of objects or variables (engineering unit, version number, etc.), there are special services for fast access, properties are leafs of each hierarchy|

I don't think this is drawn directly from the specification, but since I don't have a copy of the specs this is as good as I could find evidence.

Whitepaper link: http://www.ascolab.com/images/stories/ascolab/doc/ua_whitepaper_technicaloverview_e.pdf

closed 2018-03-29 by w4tsn