FreeOpcUa/opcua-modeler #57: Unexpected (problematic?) XML output of data variables 'value' components

w4tsn, 2018-04-01:
When I create a variable of any underlying data type, within the saved XML the format is as follows:

```
<Value>
   <uax:Double>0.0</uax:Double>
</Value>
```

I'd expect something more like the following, so that an interpreter is able to create the correct value object component under a variable type object. See the [example of the open62541 XML node-set compiler](https://open62541.org/doc/0.3/nodeset_compiler.html) for reference.

```
<Value>
    <ListOfExtensionObject>
        <ExtensionObject>
            <TypeId>
                <Identifier>i=63</Identifier>
            </TypeId>
            <Body>
                <Argument>
                    <Name>stopped</Name>
                    <DataType>
                        <Identifier>i=11</Identifier>
                    </DataType>
                    <ValueRank>-1</ValueRank>
                    <ArrayDimensions></ArrayDimensions>
                    <Description/>
                </Argument>
            </Body>
        </ExtensionObject>
    </ListOfExtensionObject>
</Value>
```

[This issue in node-red-contrib-iiot-opcua](https://github.com/biancode/node-red-contrib-iiot-opcua/issues/47) is the underlying use-case problem. Basically a node-set compiler of the node-opcua is not able to interpret the xml file correctly, which leads to the 'value' component not being created thus no read/write operations are possible. I'm not quite sure yet, if this is a problem with the modeler or the node-red or node-opcua projects.

w4tsn, 2018-04-01:
I noticed, that this seems to occur for Float and Double but not for String and Boolean.

Also the provided alternative XML schema from open62541 does produce a working result as a work-around, but produces an ugly structure of `ExtensionObject` and `List of ExtensionObject` which resolves after the first Double/Float value is written to this `value` field. I'm definitely abusing stuff here :sweat_smile:

oroulet, 2018-04-01:
The two examples you show are completely different. One is a double, the other a list of extension object... Can you check that? Uamodeler from unified automation is usually the reference. Can you write an example there and the same with opcua-modeler and we can compare

oroulet, 2018-04-01:
Btw the example above is a list of arguments. This is used for methods and is quite special

w4tsn, 2018-04-01:
Ok, I was quite confused here. It may be a problem at the node-red side since it is only happening for Double and Float but these are using the same XML schema than String and Boolean which work.

I'll check that first.

w4tsn, 2018-04-18:
The issue was caused by the node-opcua server xml loader and is fixed in pull request https://github.com/node-opcua/node-opcua/pull/476

closed 2018-04-18 by w4tsn

oroulet, 2018-04-26:
Good! so far it has always been the other projects that had errors 😂🍾