FreeOpcUa/opcua-modeler #59: How to connect to local opc server?

WangAllen, 2018-04-16:
if i have an opc Server,how could i make them connect,maybe this question can be interpreted as where in the code "opcua-modeler" could i change to make it communicate with my local server, i provide local server's address or the code could identified the address of server by its own?

zerox1212, 2018-04-16:
OPC UA modeler isn't a OPC UA client. It can't connect to another server.

I'm assuming you want to connect to an existing OPC UA server and export the address space as XML?

zerox1212, 2018-04-17:
Modeler is just for building OPC UA address space XML files.

You should use OPC UA client to talk to your server. You can use https://github.com/FreeOpcUa/opcua-client-gui or UA Expert (https://www.unified-automation.com/products/development-tools/uaexpert.html).