FreeOpcUa/opcua-modeler #60: Odd behavour when adding XML-defined objects

kleskjr, 2018-04-26:
Hi,

I stumbled upon some odd behaviour while trying to understand how to use XML Schemas. I used the myNS.xml example from [open62541 docs](https://open62541.org/doc/current/nodeset_compiler.html#getting-started). 

So the opcua-modeler shows the object definitionas as expected (in Types/Object Types):

[image]

In the definition, _Pump_ is a child of _FieldDevice_. However, if I add a new object (in Objects/a folder) as defined above, then unexpected results happen:
[image]

The _device_ object has just two subnodes (_ManufacturerName_ and _ModelName_), with no _pump_ in it. Then if _pump_ is added, it has all its subnodes plus _ManufacturerName_ and _ModelName_. The last two variables are actually siblings of the _pump_ and should not get as children in the _pump_ description.

So the question is whether it is desired behaviour, there is something funny in the XML description, or whether there is a bug in the modeler handling objects?

cheers, Nikolay

oroulet, 2018-04-26:
I am not entirely sure to understand but if you create a new object of type pump it is expected that the pump object also contains ManufacturerName and ModelName since the Pump object inherits FieldDevice. 
Id you create an object of type FieldDevice, it should only have two properties.

But what is strange in your screenshots is that the properties have a T on them, that means that they are types....I think they should not.... might be a bug at open62541 tooo

kleskjr, 2018-04-26:
thanks, oroulet! I wasn't aware that the subnodes in Objects Type mean inheritance. The fact that variables remain type is a separate issue. I will have a look and will open a new issue if it comes from the modeler.

zerox1212, 2018-04-26:
Can you start with a brand new address space in modeler and hand build these types. Build the pump type, then build the device type and put the pump inside it. then instantiate the device type. I do not really agree with @oroulet as I think you should get a copy of your device type structure, not the strange inheritance you are showing. I suspect you XML has extra references or something.

kleskjr, 2018-04-26:
@zerox1212, in a new address space, the results are similar. A pump objects inherits the properties of the device ObjectType, and a device object doesn't consist of any pumps. If I create another Device2 ObjectType, and put a Pump Object in, the device2 object still has no pump in it.

kleskjr, 2018-04-26:
Are the variables that constitute the ObjectType supposed to be of type Variable or DataType? In the open62541 example the names are DataTypes, while in opcua-modeler, I could set only variables.

zerox1212, 2018-04-26:
Interesting. I'll have to dig up my old projects and test it. I don't remember that happening, but I haven't been working with UA for a long time.

oroulet, 2018-04-27:
@kleskjr i am quite sure the open62541 example is broken. Look at other object typed in standard address space and open bug report for them