FreeOpcUa/opcua-modeler #63: Auto NodeID assigns node IDs used by the default UA namespace

mmcc-xx, 2018-06-21:
Starting from a new model, when I create a new object type and check the Auto NodeID button, the node ID that is assigned is 20001. Subsequent auto node IDs increment up from there. 

20001 is already used by PubSubDiagnosticsDataSetWriterType_Counters_StateOperationalFromError_DiagnosticsLevel according to https://github.com/OPCFoundation/UA-Nodeset/blob/master/Schema/NodeIds.csv

oroulet, 2018-06-21:
You should probably user your own namespace anyway.... Maybe it should just start at 1.... And maybe refuse to create node with namespace 0? Any idea?

zerox1212, 2018-06-21:
The pub sub part of the spec was only recently added so this wasn't an issue before. However I'm not sure what a good starting point for Auto NodeID is.

mmcc-xx, 2018-06-21:
I added a new namespace to the namespace array and made my type part of that namespace - that seems to have worked.