FreeOpcUa/opcua-modeler #66: Browsing for modeling rules?

mmcc-xx, 2018-08-17:
I'm creating a type that contains an object that has a modeling rule of mandatory placeholder.
When adding the reference, I was able to find the "HasModellingRule" reference type easily enough.

However, when attempting to assign the node ID that represented MandatoryPlaceholder, I couldn't find it by browsing. I ended up searching through the OPC UA NodeSet XML file to figure out that I wanted node ID 11510.

Is a way, within the modeling tool, for finding the appropriate node IDs associated with the HasModellingRule reference type?

oroulet, 2018-08-18:
It is somewhere... i remember having trouble to find them. Are you sure you need to create it? It should be created automatically...

oroulet, 2018-08-18:
But we really should create a folder by reference type to make this easier.

mmcc-xx, 2018-08-18:
> Are you sure you need to create it? It should be created automatically...
When I add an object to a type definition, there is no HasModellingRule reference created automatically, if that is what you are asking.

Some context awareness when adding the references would be great! One of the reasons I like this tool is that it exposes the references, making it less mysterious how the various nodes are related to each other.

oroulet, 2018-08-19:
I see the change I was talking about is not released yet... can you test master? I am doing a bit more testing and releasing soon

oroulet, 2018-08-21:
try to update to last version. I did not fix your issue, but we create some links automatically

mmcc-xx, 2018-08-28:
It looks like when I add an object to a type, it now has "ModellingRule_Mandatory" assigned automatically. Thanks!

ccibis, 2021-11-05:
I don't know, whether it fits here, but I'd like to assign the ModelingRule "optional" to a variable or an ObjectType which is a component of another object. Can anyone tell me, how to do so? Thanks in advance!

Best regards,

Christoph