FreeOpcUa/opcua-modeler #69: Install opcua-modeler on MacOS

mw84367, 2018-08-21:
Hi,

I've installed opcua-modeler on MacOS using the way it is done on Linux.

`pip install opcua-modeler
Collecting opcua-modeler
Requirement already satisfied: opcua-widgets>=0.5.5 in /usr/local/lib/python2.7/site-packages (from opcua-modeler) (0.5.5)
Requirement already satisfied: opcua>=0.95.1 in /usr/local/lib/python2.7/site-packages (from opcua-modeler) (0.98.5)
Requirement already satisfied: lxml in /usr/local/lib/python2.7/site-packages (from opcua>=0.95.1->opcua-modeler) (4.2.4)
Requirement already satisfied: enum34 in /usr/local/lib/python2.7/site-packages (from opcua>=0.95.1->opcua-modeler) (1.1.6)
Requirement already satisfied: trollius in /usr/local/lib/python2.7/site-packages (from opcua>=0.95.1->opcua-modeler) (2.2)
Requirement already satisfied: pytz in /usr/local/lib/python2.7/site-packages (from opcua>=0.95.1->opcua-modeler) (2018.5)
Requirement already satisfied: python-dateutil in /usr/local/lib/python2.7/site-packages (from opcua>=0.95.1->opcua-modeler) (2.7.3)
Requirement already satisfied: futures in /usr/local/lib/python2.7/site-packages (from opcua>=0.95.1->opcua-modeler) (3.2.0)
Requirement already satisfied: six in /usr/local/lib/python2.7/site-packages (from trollius->opcua>=0.95.1->opcua-modeler) (1.11.0)
Installing collected packages: opcua-modeler
Successfully installed opcua-modeler-0.5.5
`

When I run opcua-modeler I get the following error:

`host:~ user$ opcua-modeler
No handlers could be found for logger "opcua.client.client"
Traceback (most recent call last):
  File "/usr/local/bin/opcua-modeler", line 7, in <module>
    from uamodeler.uamodeler import main
  File "/usr/local/lib/python2.7/site-packages/uamodeler/uamodeler.py", line 21, in <module>
    from uamodeler.uamodeler_ui import Ui_UaModeler
ImportError: No module named uamodeler_ui`

It looks like that there is something wrong with paths.

How can I fix it?

Thanks

Mike

oroulet, 2018-08-21:
You need Python 3 not Python 2

mw84367, 2018-08-21:
Thanks for your very quick response.

This fixed the issue ;)