FreeOpcUa/opcua-modeler #71: opcua modeler problem

ericccccccc, 2018-10-04:
hi~i'm start to leran about opcau open source.
i want to exec opcua-modeler, but i got this error ~

[image]

oroulet, 2018-10-04:
Your Python version is too old..

ericccccccc, 2018-10-04:
thx for ur suggestion~ i'm also learning about python 2.
now i have update my python version to 3.6 , but i got another problems : 
 ImportError: cannot import name 'objectify'

[image]

thx for your help!!

ericccccccc, 2018-10-04:
i have fix that by install the new lxml version !
thx !!

closed 2018-11-07 by oroulet