FreeOpcUa/opcua-modeler #74: Can't add BaseDataVariableType to a BaseObjectType

the-grandson, 2018-10-22:
If you right click on a `BaseObjectType` and next click `Add Variable` you are only allowed to add `BaseDataType`. As a way around currently you are allowed to do it using the `References Editor`.

the-grandson, 2018-10-24:
Sorry, I didn't explained correctly. In fact I can't add a `DataVariable` to an `BaseObjectType`.

You can check for example the `StateMachineType (NodeId i=2299)`. It's a `BaseObjectType` from OPC UA Node Set, and it has a variable `CurrentState (NodeId i=2769)`. That variable has type definition `StateVariableType (NodeId i=2755)`, which is a `BaseDataVariableType`.

I can't do the same with my custom `BaseObjectType's`. The window just let you add `BaseDataType's`.

oroulet, 2018-11-07:
sorry forgot that bug report. 
StateMachineState is anew object type and has a variable yes. The same way you can make a new object type and add a variable to it. This is no problem and works. What is a DataVariable?

closed 2019-02-01 by oroulet