FreeOpcUa/opcua-modeler #75: Error defining a `Property` data type as `Guid`

the-grandson, 2018-10-22:
Can't define a `Property` data type as `Guid`, it throws the following error:

`uawidgets.utils - ERROR - badly formed hexadecimal UUID string')
Traceback (most recent call last):
  File "C:\User\python-3.6.6.amd64\lib\site-packages\uawidgets\utils.py", line 21, in wrapper
    result = func(self, *args)
  File "C:\User\python-3.6.6.amd64\lib\site-packages\uamodeler\uamodeler.py", line 324, in add_variable
    args, ok = NewUaVariableDialog.getArgs(self.modeler, "Add Variable", self._model_mgr.server_mgr, default_value=9.99)
  File "C:\User\python-3.6.6.amd64\lib\site-packages\uawidgets\new_node_dialogs.py", line 94, in getArgs
    return dialog.get_args(), True
  File "C:\User\python-3.6.6.amd64\lib\site-packages\uawidgets\new_node_dialogs.py", line 163, in get_args
    var = string_to_variant(self.valLineEdit.text(), vtype)
  File "C:\User\python-3.6.6.amd64\lib\site-packages\opcua\common\ua_utils.py", line 122, in string_to_variant
    return ua.Variant(string_to_val(string, vtype), vtype)
  File "C:\User\python-3.6.6.amd64\lib\site-packages\opcua\common\ua_utils.py", line 111, in string_to_val
    val = uuid.UUID(string)
  File "C:\User\python-3.6.6.amd64\lib\uuid.py", line 140, in __init__
    raise ValueError('badly formed hexadecimal UUID string')
ValueError: badly formed hexadecimal UUID string`

zerox1212, 2018-10-22:
I don't remember exactly how the window looks, does it let you enter your own UUID value?

the-grandson, 2018-10-22:
No, the window that pop's up is the same as in the other DataTypes, check
below.
[image: GUID.PNG]

zerox1212, 2018-10-23:
Your screenshot didn't work.

I have used UUID before so I know it used to work. Maybe a bug was introduced recently because not many people use UUID type.

the-grandson, 2018-10-24:
I've modified the XML file manually and filled the property with a UUID.
Afterwards import the modified model and the modeler works just fine.
Maybe is just a problem generating a new UUID when you click 'Ok'.

oroulet, 2018-10-24:
the convertion is done in ua_utils.py", line 111, in string_to_val maybe you should have a look there. maybe it gets a string and want bytes or something else

oroulet, 2019-02-01:
has anybody an example of an uuid that should works? I have noe idea since I never use them

the-grandson, 2019-02-01:
This one `59f1dbe6-90b8-11e8-9eb6-529269fb1459` I am using, it should work.
Added manually to the XML file after exporting, and imported again with no
trouble.

oroulet, 2019-02-01:
this works fine here... I do not know if that got fixed before... but I pushed a change to master so now a guuid is created automatically as defualt value. make things easier for testing, etc..

closed 2019-02-01 by oroulet

zerox1212, 2019-02-01:
It could be that it never generated an UUID automatically?