FreeOpcUa/opcua-modeler #76: Deletion of nodes causes BadNodeIdUnknown error when trying to save as XML

dwolfschlaeger, 2019-01-31:
Dear all, 
working the first time with the modeler I noticed, that the deletion of redundant nodes causes a crash of the full model. 
It seems to me, that internally the deleted nodes are still stored somewhere. When trying to save the list of new nodes, the information is no longer available and evoke the BadNodeIdUnknown error message.

Steps to reproduce:

1. Add a new ObjectType Node under BaseObjectTypes.
2. Add a new VariableNode as component of the Objecttype node. 
3. Delete the ObjectType Node.
4. Try saving the model.

I'd be happy if someone could explain to me, whether this behaviour is intended and how it is possble to debug the model anyhow in that case. Otherwise the full model would need to build from scratch. 
Thanks!

zerox1212, 2019-01-31:
I don't have time to look at this, but I can say that deletion of nodes has to be done carefully. It is not very intelligent from a cross referencing standpoint. For example if you make an object type, then create a node that uses that object type, then edit the object type again, your created node won't be updated.

I'm guessing the same applies for deletion. That said you will most likely need to clone this library and run the app while debugging so you can see what reference it's crashing on.

oroulet, 2019-01-31:
Will have a look. Nice if you can report as much as possible. There are many small bugs around we need to find and fix

oroulet, 2019-01-31:
Are you running master btw?

dwolfschlaeger, 2019-01-31:
I agree that the deletion of nodes must be performed carefully. Nevertheless, sometimes one mistakingly creates a node or decides that the modeling was not optimal, so changes should be possible. 
Maybe the possibility to undo the previous operation could be an improvement though?

No, I am currently running version 0.5.5 installed with pip.

oroulet, 2019-02-01:
Hi, I fixed it on master. but you must run master version of python-opcua, opcua.modeller and uawidget.
This used to workbut you know... api changes...
We should add tests for it, testing UI applications is not so easy..

oroulet, 2019-02-01:
btw undo is much harder to implement, you need to keep track of entire history, etc... so it is much easier to have a good delete ;-)

oroulet, 2019-02-01:
thank you for the bug report and do not hesitate to report more!

oroulet, 2019-02-01:
OK released everythin. upgrade with
pip install opcua-modeller --upgrade