FreeOpcUa/opcua-modeler #78: [0.5.6] Rightclick on Attribute leads to crash

araisch, 2019-02-18:
Create a Variable, Rightclick anything in "Attributes Editor" -> Crash

```
uamodeler.model_manager - INFO - Creating variable with args: (TwoByteNodeId(i=0), QualifiedName(0:NoName), Variant(val:False,type:VariantType.Boolean), <VariantType.Boolean: 1>, NumericNodeId(i=1))')
Traceback (most recent call last):
  File "c:\program files (x86)\wpy-3710\python-3.7.1.amd64\lib\site-packages\uawidgets\attrs_widget.py", line 130, in showContextMenu
    item = self.get_current_item()
  File "c:\program files (x86)\wpy-3710\python-3.7.1.amd64\lib\site-packages\uawidgets\attrs_widget.py", line 136, in get_current_item
    idx = idx.siblingAtColumn(col_idx)
AttributeError: 'QModelIndex' object has no attribute 'siblingAtColumn'
```

oroulet, 2019-02-18:
you  have some old qt version. you are welcome to find out how to do that with old qt version and make a PR

araisch, 2019-02-18:
Thanks a lot for the hint. Thought I got the latest version, due to installing everything today.
If someone else encounter that:
`pip install --upgrade pyqt5`

closed 2019-02-18 by araisch