FreeOpcUa/opcua-modeler #81: Debian stretch / opcua-modeler fails to start / recent files

darxridge, 2019-04-10:
System: python3.7

Stack dump:
Traceback (most recent call last):
  File "app.py", line 3, in <module>
    main()
  File "/home/rich/src/opcua-modeler/opcua-modeler/uamodeler/uamodeler.py", line 544, in main
    modeler = UaModeler()
  File "/home/rich/src/opcua-modeler/opcua-modeler/uamodeler/uamodeler.py", line 407, in __init__
    self._update_recent_files_ui()
  File "/home/rich/src/opcua-modeler/opcua-modeler/uamodeler/uamodeler.py", line 428, in _update_recent_files_ui
    for idx, path in enumerate(self._recent_files):
TypeError: 'NoneType' object is not iterable
Makefile:6: recipe for target 'run' failed
make: *** [run] Error 1

Suggestion: uamodeler.py[428]
  for idx, path in enumerate(self._recent_files or []):

oroulet, 2019-04-10:
thanks. Fixed in master just now (I did no see your suggestion but it was good ;-) )

oroulet, 2019-04-10:
feel free to fix other small things like this one