FreeOpcUa/opcua-modeler #83: Ubuntu 16.04 / opcua-modeler fails to start after installing

6guenther, 2019-06-12:
I am not quite sure what my problem is here, tried installing the modeler using `pip3 install opcua-modeler` and got the following error message:
`Traceback (most recent call last):
  File "/home/ros/.local/bin/opcua-modeler", line 7, in <module>
    from uamodeler.uamodeler import main
  File "/home/ros/.local/lib/python3.5/site-packages/uamodeler/uamodeler.py", line 15, in <module>
    from uawidgets.attrs_widget import AttrsWidget
  File "/home/ros/.local/lib/python3.5/site-packages/uawidgets/attrs_widget.py", line 11, in <module>
    from uawidgets.get_node_dialog import GetNodeButton
  File "/home/ros/.local/lib/python3.5/site-packages/uawidgets/get_node_dialog.py", line 7, in <module>
    from uawidgets.tree_widget import TreeWidget
  File "/home/ros/.local/lib/python3.5/site-packages/uawidgets/tree_widget.py", line 62
    raise ValueError(f"Node {node} not found in tree")`

OS is Ubuntu 16.04 
PyQt5 is installed

zerox1212, 2019-06-12:
You need Python 3.6 or higher. You are using 3.5.

6guenther, 2019-06-12:
Fixed it, thanks for the quick reply!

closed 2019-06-12 by 6guenther

oroulet, 2019-06-12:
Can you make a pull request?

zerox1212, 2019-06-12:
I edited the ReadMe to clarify this point. It has been in the Issues a few times.