FreeOpcUa/opcua-modeler #84: can't import new offical nodeset xml

flyingToast3r, 2019-07-31:
I tried to import the nodeset file [Opc.Ua.Di.NodeSet2.xml](https://opcfoundation.org/UA/schemas/DI/1.2/Opc.Ua.Di.NodeSet2.xml) and got the error:
`opcua.common.xmlimporter - WARNING - failure adding node NodeData(nodeid:NumericNodeId(ns=1;i=15035))')
uawidgets.utils - ERROR - The parent node id does not to refer to a valid node.(BadParentNodeIdInvalid)')`

Then I tried the older version 1.0 file as stated in the Master Plan under version 0.3:[Opc.Ua.Di.NodeSet2.xml](https://opcfoundation.org/UA/schemas/DI/1.0/Opc.Ua.Di.NodeSet2.xml ) and that worked. 

With other new nodeset files like from the Robotics I have the same error as stated above: [Opc.Ua.Robotics.NodeSet2.xml](https://opcfoundation.org/UA/schemas/Robotics/1.0/Opc.Ua.Robotics.NodeSet2.xml)

I couldn't track down the issue for the moment. Does anyone have some suggestions?
I used the latest relese version as well as the bleeding edge developer version from the master branch.

oroulet, 2019-08-07:
that would be really nice to fix. it probably some issues in the xml importer of python-opcua

zerox1212, 2019-08-07:
XML handling is very complex, you will have to spend some time trying to debug it. Maybe you can remove most of the NodeSet2 so that you can look only at the broken node.

oroulet, 2020-03-25:
OK I found out that a newer version of the standard node definiton is needed, but they seem to have changed a litle the format :-1:

JMayrbaeurl, 2020-07-02:
@oroulet We've run into the same issue when trying to open version 1.0.4 of the official Robotics nodesets file. This file seems to be pretty current, since the last change was 6 days ago. What's needed to be able to open the Robotics nodesets file?