FreeOpcUa/opcua-modeler #85: Nodeset import error

Lukas1811, 2019-08-19:
There are a few different bugs while importing nodesets, in my case it is the official DI nodeset by the OPC foundation. The first problem is that it only works if I import the Opc.Ua.Di.Nodeset2.xml file from the python-opcua schema folder, if I import the one from the OPC Doundations GitHub repo I get a lot of errors.
The next issue is that if I save the file and reload it, the programm couldn't reload the previously imported Nodeset, due to the fact that only the file name is saved in the .uamodel file and not the path. If you add the path each time manually the import works flawless.

oroulet, 2019-08-19:
I known there are many rough edges in that progam. Great if you could try to fix this. The issue with saving path is that you cannot move the file afterward... not sure how to solve that...

zerox1212, 2019-08-20:
Saving path might be better. If someone moves it we could throw a file not found error.

As for other bugs, keep in mind you also need to look at code in the main python opcua library. That is where all the XML stuff is.

oroulet, 2019-08-20:
maybe we can save the path but if the file name is found in the same directory then use it

benkinooby, 2020-03-05:
How about using prioritized "include paths" where the modeler searches for the referenced model files. The paths may be absolute as well as relative paths. By default, the current path ``./`` has highest priority (this is what is proposed in https://github.com/FreeOpcUa/opcua-modeler/issues/85#issuecomment-523149467 with optional, additional paths to search)