FreeOpcUa/opcua-modeler #87: TypeError: 'NoneType' object is not iterable

parth-ifam, 2019-08-20:
I am just starting the opcua-modeler. I have python 3.6.* and PyQt5. It actually started the first time. But now i am getting the following error:
```
ubuntu1604@ubuntu1604-VirtualBox:~$ opcua-modeler
cryptography is not installed, use of crypto disabled
cryptography is not installed, use of crypto disabled
cryptography is not installed, use of crypto disabled
Traceback (most recent call last):
  File "/home/ubuntu1604/.local/bin/opcua-modeler", line 11, in <module>
    sys.exit(main())
  File "/home/ubuntu1604/.local/lib/python3.6/site-packages/uamodeler/uamodeler.py", line 544, in main
    modeler = UaModeler()
  File "/home/ubuntu1604/.local/lib/python3.6/site-packages/uamodeler/uamodeler.py", line 407, in __init__
    self._update_recent_files_ui()
  File "/home/ubuntu1604/.local/lib/python3.6/site-packages/uamodeler/uamodeler.py", line 428, in _update_recent_files_ui
    for idx, path in enumerate(self._recent_files):
TypeError: 'NoneType' object is not iterable
```
Dont know how to proceed further.

oroulet, 2019-08-20:
This is supposed to be fixed. Can you try master?

parth-ifam, 2019-08-20:
Thanks for a quick reply, what do you mean by master ? Is there a different git branch ?

oroulet, 2019-08-20:
The problem is that we initialized list to None instead of empty if not config value is found ( new user case). I am sure someone made a patch for it...  Should be in master branch at least

parth-ifam, 2019-08-20:
Okay , i solved the problem myself by adding a line of code as described to the file `~/.local/lib/python3.6/site-packages/uamodeler/uamodeler.py`
Before it was:
```
    def _update_recent_files_ui(self):
        for idx, path in enumerate(self._recent_files):
            self._recent_files_acts[idx].setText(path)
            self._recent_files_acts[idx].setData(path)
            self._recent_files_acts[idx].setVisible(True)
```
and after adding a line:
```
    def _update_recent_files_ui(self):
        if self._recent_files is not None:
            for idx, path in enumerate(self._recent_files):
                self._recent_files_acts[idx].setText(path)
                self._recent_files_acts[idx].setData(path)
                self._recent_files_acts[idx].setVisible(True)
```

closed 2019-08-20 by parth-ifam

oroulet, 2019-08-20:
this looks fine, could you make a PR? thanks