FreeOpcUa/opcua-modeler #93: improvement: use OPC Foundation UANodeSet files

benkinooby, 2020-03-05:
The OPC Foundation is maintaining a collection of UANodeSet files at
https://github.com/OPCFoundation/UA-Nodeset

It would be helpful to include those as git submodule to make them easily available to the user.

Related (Correct me if I am wrong):
When creating a new model, the standard UANodeSet Opc.UA.NodeSet2.xml visible in the sub-window "Node Sets" seems to be generated with python code from `opcua/server/standard_address_space/standard_address_space_part5.py` instead of being loaded from a nodeset file. Consequently, only models that happen to be compatible to that version (whatever that version is, I did not dig long enough to find out) can be successfully loaded with Free OPC UA modeler. Correct?

oroulet, 2020-03-05:
Yes we use these nodesets. But for the server it was faster to generate python code from them instead of importing everything from xml every time. They seem to be backwards compatible so maybe we can just keep them up to date

benkinooby, 2020-03-05:
Thank you very much for your patient reply.

As reference for myself and others reading this:
Are you referring to the following files?
https://github.com/FreeOpcUa/python-opcua/tree/master/schemas

One thing I noticed on a quick glance is, that "your" Opc.Ua.Di.NodeSet2.xml is missing the ``<Models>`` element (both of them are on master branch, so the comparison may be moot)

Yours:
https://github.com/FreeOpcUa/python-opcua/blob/6ce90cea31e44a036d5030552b9b10badab738e0/schemas/Opc.Ua.Di.NodeSet2.xml#L34

Theirs:
https://github.com/OPCFoundation/UA-Nodeset/blob/ad205e664f88d3989cb06671207ca3f8932db06c/DI/Opc.Ua.Di.NodeSet2.xml#L34

andreafresa, 2020-04-23:
hello Guys, I am trying to import the Robotics section of OpcUa but I always get the same issue: 
"The parent node id does not to refer to a valid node.(BadParentNodeIdInvalid)".
Do you know how to fix it?

oroulet, 2020-04-23:
We need to update the standard namespace in python-opcua. unfortunately they changed the format a little and we need to fix something deep in parser. See https://github.com/FreeOpcUa/python-opcua/issues/1000

and https://github.com/FreeOpcUa/opcua-asyncio/pull/173

oroulet, 2020-04-23:
hep welcome

oroulet, 2020-05-02:
should be fixed now. added last nodeset in both pyton-opcu and  opcua-asyncio

andreafresa, 2020-05-06:
Hey guys, 
I am trying again to import the Robotics Nodeset on opcua-modeler (Actions->Import XML). 
I always get the issue I posted in the comment above. 
I am on windows and I tried to update the opcua-modeler using "pip install opcua-modeler --upgrade" from my winpython environment.
Is this bug solved only in the dev version (the one that must be cloned from git)?

JMayrbaeurl, 2020-07-02:
Just tried it with https://github.com/OPCFoundation/UA-Nodeset/blob/v1.04/Robotics/Opc.Ua.Robotics.NodeSet2.xml in the 'package' and in the 'Dev version'. Always gives the same error:

`opcua.common.xmlimporter - WARNING - failure adding node NodeData(nodeid:NumericNodeId(ns=1;i=1002))')
uawidgets.utils - ERROR - "The parent node id does not to refer to a valid node."(BadParentNodeIdInvalid)')
Traceback (most recent call last):
  File "C:\Dev\nebula\opcua\modeler\opcua-widgets\uawidgets\utils.py", line 21, in wrapper
    result = func(self, *args)
  File "c:\Dev\nebula\opcua\modeler\opcua-modeler\uamodeler\uamodeler.py", line 249, in open
    self._model_mgr.open(path)
  File "c:\Dev\nebula\opcua\modeler\opcua-modeler\uamodeler\model_manager.py", line 181, in open
    self.open_xml(path)
  File "c:\Dev\nebula\opcua\modeler\opcua-modeler\uamodeler\model_manager.py", line 113, in open_xml
    self._open_xml(path)
  File "c:\Dev\nebula\opcua\modeler\opcua-modeler\uamodeler\model_manager.py", line 119, in _open_xml
    path = self.import_xml(path)
  File "c:\Dev\nebula\opcua\modeler\opcua-modeler\uamodeler\model_manager.py", line 102, in import_xml
    new_nodes = self.server_mgr.import_xml(path)
  File "c:\Dev\nebula\opcua\modeler\opcua-modeler\uamodeler\server_manager.py", line 68, in import_xml
    return self._backend.import_xml(path)
  File "c:\Dev\nebula\opcua\modeler\opcua-modeler\uamodeler\server_manager.py", line 112, in import_xml
    return self._server.import_xml(path)
  File "C:\Dev\nebula\opcua\modeler\python-opcua\opcua\server\server.py", line 510, in import_xml
    return importer.import_xml(path, xmlstring)
  File "C:\Dev\nebula\opcua\modeler\python-opcua\opcua\common\xmlimporter.py", line 67, in import_xml
    node = self._add_node_data(nodedata)
  File "C:\Dev\nebula\opcua\modeler\python-opcua\opcua\common\xmlimporter.py", line 84, in _add_node_data
    node = self.add_object_type(nodedata)
  File "C:\Dev\nebula\opcua\modeler\python-opcua\opcua\common\xmlimporter.py", line 196, in add_object_type
    res[0].StatusCode.check()
  File "C:\Dev\nebula\opcua\modeler\python-opcua\opcua\ua\uatypes.py", line 218, in check
    raise UaStatusCodeError(self.value)
opcua.ua.uaerrors._auto.BadParentNodeIdInvalid: "The parent node id does not to refer to a valid node."(BadParentNodeIdInvalid)`

oroulet, 2020-07-04:
strange I just tested the examples in opcua-asyncio and python-opcua and the robot nodeset seems to import fine. Can you update opcua and try to run example at: https://github.com/FreeOpcUa/python-opcua/blob/master/examples/server-import-robot-nodeset.py

JMayrbaeurl, 2020-07-06:
Problem seems to come from not correctly loading the required model dependency on OPC-UA DI. Form the Robotics IM file:

  <NamespaceUris>
    <Uri>http://opcfoundation.org/UA/Robotics/</Uri>
    <Uri>http://opcfoundation.org/UA/DI/</Uri>
  </NamespaceUris>
  <Models>
    <Model ModelUri="http://opcfoundation.org/UA/Robotics/" Version="1.0.0" PublicationDate="2019-05-06T00:00:00Z">
      <RequiredModel ModelUri="http://opcfoundation.org/UA/" Version="1.04" PublicationDate="2019-05-01T00:00:00Z" />
      <RequiredModel ModelUri="http://opcfoundation.org/UA/DI/" Version="1.02" PublicationDate="2019-05-01T00:00:00Z" />
    </Model>
  </Models>

Namespace 2 is for DI and the error on importing the Robotics file states an invalid parent node id. 

<UAObjectType NodeId="ns=1;i=1002" BrowseName="1:MotionDeviceSystemType">
    <DisplayName>MotionDeviceSystemType</DisplayName>
    <Description>Contains the set of controllers and motion devices in a closely-coupled motion device system.</Description>
    <Documentation>https://reference.opcfoundation.org/v104/Robotics/v100/docs/7.1.2</Documentation>
    <References>
      <Reference ReferenceType="HasComponent">ns=1;i=5002</Reference>
      <Reference ReferenceType="HasComponent">ns=1;i=5001</Reference>
      <Reference ReferenceType="HasComponent">ns=1;i=5010</Reference>
      <Reference ReferenceType="HasSubtype" IsForward="false">ns=2;i=15063</Reference>
    </References>
  </UAObjectType>

Parent ns=2;i=15063 is the ComponentType from DI, that's obvisouly not loaded

oroulet, 2020-07-06:
@JMayrbaeurl what do you mean by dev version?
when I import Di and robotics xl files here with python-opcya master https://github.com/FreeOpcUa/python-opcua/blob/master/examples/server-import-robot-nodeset.py I do not see any warnings (but setting logging to INFO I get a looooooot of warnings)
Helps is anyway welcome to improve xml parser

JMayrbaeurl, 2020-07-06:
@oroulet Finally found a working setup with the modeler:
1) Open modeler application and select menu item 'New model'
2) Import DI nodes set xml file
3) Import Robotics nodes set xml file

The problem or issue is that opening a nodes set xml file, that depends on other nodes set xml files as the Robotics has with DI, doesn't automatically open and parse the depending xml files, causing errors when referencing nodes from them.

oroulet, 2020-07-06:
Ok. Now I understand what your issue was. Not sure we can automate that. They may be at different places

JMayrbaeurl, 2020-07-06:
Can you at least ad a warning on importing a nodes set xml file that has dependencies? Something like 'Warning: This file has dependencies on .... Please import them before importing this file"

oroulet, 2020-07-06:
good point..

pbertoni89, 2020-11-17:
Got a similar problem (I just opened [python-opcua #1166](https://github.com/FreeOpcUa/python-opcua/issues/1166)) while importing PackML official NodeSet.
 @zerox1212 perhaps this a more appropriate thread, sorry for the little overlapping.
Current PackML says as the only requirement
```
<Model ModelUri="http://opcfoundation.org/UA/PackML/" Version="1.1.0" PublicationDate="2020-05-28T08:03:15Z">
    <RequiredModel ModelUri="http://opcfoundation.org/UA/" Version="1.04.3" PublicationDate="2019-09-09T00:00:00Z" />
</Model>
```

zerox1212, 2020-11-18:
@pbertoni89 Sorry, I have never used the official node sets so I can't help you there. I always made my own custom models with modeler.

JinhwanPark, 2021-07-27:
When I try to load the CNC Nodeset, I get the following error:
TypeError: object of type 'ExtObj' has no len()
Does anyone know the cause of the following?
There seems to be a part that cannot be parsed among Nodes of CncPositionDataType
Other modelers work without issues.(ex. UA Modeler)

benkinooby, 2021-07-29:
> @pbertoni89 Sorry, I have never used the official node sets so I can't help you there. I always made my own custom models with modeler.

Hi, for reference/information:
https://opcfoundation.org/developer-tools/specifications-opc-ua-information-models/opc-ua-packml-companion-specification/

PackML is a companion specification. Companion specifications are industry-specific specifications/models developed by industrial partners in cooperation with OPC Foundation. You can find a list of companion specifications here:
https://opcfoundation.org/developer-tools/specifications-opc-ua-information-models

Adding them to the default set of models shipped with opcua-modeler would certainly be a big step up for opcua-modeler users.