FreeOpcUa/opcua-modeler #94: Can not add variables in special type

0DennisLee0, 2020-03-10:
The variable can only be added in the BaseDataType. Can not be specified in other VariableDataType like DataItem. Is is a bug?

araisch, 2020-07-28:
Maybe related:

1. Adding a new Variable as Component of an ObjectType with the PopUp (Rightclick -> Add Variable)
2. Choose the TypeDefinition (e.g. Boolean) -> it shows the chosen Type in the Button.
3. After clicking "OK", Output Log says `uamodeler.model_manager - INFO - Creating variable with args: (NumericNodeId(ns=2;i=0), QualifiedName(2:IsPresent), Variant(val:True,type:VariantType.Boolean), <VariantType.Boolean: 1>, NumericNodeId(i=1))')`
4. The new Reference is shown in "References Editor", but TypeDefinition column shows "BaseDataVariableType", same for the Variable itself ("HasTypeDefinition" is "BaseDataVariableType" or i=63)
5. Here you can change it to the right TypeDefinition by clicking on "NodeId" and then the 3 dots, even different from "BaseDataType", but should be the right one in first place.

Psoke, 2021-07-19:
But if you do it this way, neither the mandatory nor the optional properties of a specific type are beeing pulled. I can not find a way to add a simple VariableType like AnalogItemType directly when adding variables and either way it does not pull the property EURange with it.