FreeOpcUa/opcua-modeler #99: pip install error

kota-tsubo, 2020-05-13:
Hi,
I tried to install Free OPC UA Modeler using pip.
However I couldn't install it.
pip error is below.

_C:\WPy64-3771b3\scripts>pip install opcua-modeler
Collecting opcua-modeler
  Using cached opcua-modeler-0.5.9.tar.gz (14 kB)
Collecting opcua>=0.95.1
  Using cached opcua-0.98.10.tar.gz (454 kB)
ERROR: Could not find a version that satisfies the requirement opcua-widgets>=0.5.9 (from opcua-modeler) (from versions: 0.1.0, 0.2, 0.3.0, 0.4.0, 0.5.0, 0.5.1, 0.5.2, 0.5.3, 0.5.4, 0.5.5, 0.5.6, 0.5.7, 0.5.8)
ERROR: No matching distribution found for opcua-widgets>=0.5.9 (from opcua-modeler)_

How can I install it successfully?

oroulet, 2020-05-13:
thanks for the report. I releasedanew version. Should worknow

closed 2020-05-13 by oroulet

halitekmekcioglu, 2024-03-21:
Hello 
 
I get same pip error 

WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProxyError('Cannot connect to proxy.', NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x000002533AEAA790>: Failed to establish a new connection: [WinError 10061] No connection could be made because the target machine actively refused it'))': /artifactory/api/pypi/python-virtual/simple/opcua/
ERROR: Could not find a version that satisfies the requirement opcua (from versions: none)
ERROR: No matching distribution found for opcua