FreeOpcUa/opcua-modeler #100: Can't import S7-1500 XML

Ogoku, 2020-05-19:
Hi,

when I try to import the XML that I have exportet via the TIA Portal (V15.1) the Modeler throws this Error:

uawidgets.utils - ERROR - "The parent node id does not to refer to a valid node."(BadParentNodeIdInvalid)')
Traceback (most recent call last):
  File "C:\Users\ogoreks\AppData\Roaming\Python\Python38\site-packages\uawidgets\utils.py", line 21, in wrapper
    result = func(self, *args)
  File "C:\Users\ogoreks\AppData\Roaming\Python\Python38\site-packages\uamodeler\uamodeler.py", line 262, in import_xml
    self._model_mgr.import_xml(path)
  File "C:\Users\ogoreks\AppData\Roaming\Python\Python38\site-packages\uamodeler\model_manager.py", line 102, in import_xml
    new_nodes = self.server_mgr.import_xml(path)
  File "C:\Users\ogoreks\AppData\Roaming\Python\Python38\site-packages\uamodeler\server_manager.py", line 68, in import_xml
    return self._backend.import_xml(path)
  File "C:\Users\ogoreks\AppData\Roaming\Python\Python38\site-packages\uamodeler\server_manager.py", line 112, in import_xml
    return self._server.import_xml(path)
  File "C:\Program Files\Python38\lib\site-packages\opcua\server\server.py", line 510, in import_xml
    return importer.import_xml(path, xmlstring)
  File "C:\Program Files\Python38\lib\site-packages\opcua\common\xmlimporter.py", line 67, in import_xml
    node = self._add_node_data(nodedata)
  File "C:\Program Files\Python38\lib\site-packages\opcua\common\xmlimporter.py", line 84, in _add_node_data
    node = self.add_object_type(nodedata)
  File "C:\Program Files\Python38\lib\site-packages\opcua\common\xmlimporter.py", line 196, in add_object_type
    res[0].StatusCode.check()
  File "C:\Program Files\Python38\lib\site-packages\opcua\ua\uatypes.py", line 218, in check
    raise UaStatusCodeError(self.value)
opcua.ua.uaerrors._auto.BadParentNodeIdInvalid: "The parent node id does not to refer to a valid node."(BadParentNodeIdInvalid)

Attachet is the XML file from Siemens
[170044_Durchziehpliesten.ÜGS+SSK1-50K0.OPCUA.zip](https://github.com/FreeOpcUa/opcua-modeler/files/4648603/170044_Durchziehpliesten.UGS%2BSSK1-50K0.OPCUA.zip)

DaniHaag, 2020-08-05:
I have the same issues with files exported from node-opcua.
It looks like whenever there is a UAVariable that does not support the optional ParentNodeId it gets rejected.

I created a simple file to reproduce the issue:
[nodeset-simple.zip](https://github.com/FreeOpcUa/opcua-modeler/files/5028773/nodeset-simple.zip)

from https://opcfoundation.org/UA/2011/03/UANodeSet.xsd
  <xs:complexType name="UAInstance">
    <xs:complexContent>
      <xs:extension base="UANode">
        **<xs:attribute name="ParentNodeId" type="NodeId" use="optional">**</xs:attribute>
      </xs:extension>
    </xs:complexContent>
  </xs:complexType>

zerox1212, 2020-08-05:
So you are imported variable nodes with no parent?

DaniHaag, 2020-08-06:
Thanks for your quick reply.

Of course there is a parent node. I highlight this in the lines below but you will also find this in all the files attached to this issue. (this is from example-no-parent-broken.xml)

Node i=2002 has a component reference to i=2003, which represents the parent child relationship. 

but the reference is not duplicated in node i=2003 as a backward reference (@see commented out line in that node). The file works when I comment that line (@see example-no-parent-working.xml)

Just as a side node, the behaviour is even the same if it contains a parentnodeid but not the has component backward reference. (@see example-orig.xml vs example-broken.xml)

```xml
  <UAObject BrowseName="1:obj1" NodeId="ns=1;i=2002">
    <DisplayName>BaseObjectType</DisplayName>
    <Description>The base type for all object nodes.</Description>
    <References>
      <Reference IsForward="false" ReferenceType="Organizes">i=85</Reference>
      <Reference ReferenceType="HasTypeDefinition">i=58</Reference>
      <Reference ReferenceType="HasComponent">ns=1;i=2003</Reference>
      <Reference ReferenceType="HasProperty">ns=1;i=2004</Reference>
      <Reference ReferenceType="HasComponent">ns=1;i=2006</Reference>
    </References>
  </UAObject>

...

<UAVariable BrowseName="1:var1" DataType="Float" NodeId="ns=1;i=2003">
    <DisplayName>var1</DisplayName>
    <Description>var1</Description>
    <Value>
      <uax:Float>9.99</uax:Float>
    </Value>
    <References>
      <!--<Reference IsForward="false" ReferenceType="HasComponent">ns=1;i=2002</Reference>-->
      <Reference ReferenceType="HasTypeDefinition">i=63</Reference>
    </References>
  </UAVariable>
```

oroulet, 2020-08-06:
OK there was no code to handle that special case: I wrote something there: https://github.com/FreeOpcUa/opcua-asyncio/pull/251/commits/092a781761e1c46aed5ef55238d3391b26f0db35

but this is for opcua-asyncio. So you should port it to python-opcua or wait until I(or someone else) port the modeler to opcua-.asyncio one day... should not so hard though...

oroulet, 2020-08-06:
also @Ogoku I also check your siemens file. But this is a different issue. It has a node ns=2; i=1002 which does not exist so we crash on it. Must some some special siemens namespace you need to import first