FreeOpcUa/opcua-modeler #102: Pull Properties from TypeDefinition

araisch, 2020-09-22:
1. Add Variable
2. Change "HasTypeDefinition" to e.g. i=2368 (AnalogItemType)
3. Properties of the Type aren't pulled, neither Mandatory nor Optional ones.
[image]
4. Should have the mandatory properties and the inherited ones and an option to choose from the optional ones

Or is there another way to reach that?

Psoke, 2021-07-19:
THIS is exactly what I have been wondering also. In UA Modeler it simply pulls EURange with it when defining it as AnalogItemType. Can't get it to work here and its highly frustrating. 
Also why can you add a objectType when adding an object but not add a variableType when adding a variable.

zerox1212, 2021-07-20:
> THIS is exactly what I have been wondering also. In UA Modeler it simply pulls EURange with it when defining it as AnalogItemType. Can't get it to work here and its highly frustrating.
> Also why can you add a objectType when adding an object but not add a variableType when adding a variable.

Add this functionality and send a pull request.

araisch, 2021-07-21:
> Add this functionality and send a pull request.

@zerox1212 You're right: This is open source and instead of complaining aggressively anyone could fix it or add a functionality. But not everyone has the skills to do that. 
@Psoke if you just searching for a Modeler "For Free" try SIOME from Siemens. There this feature works..