FreeOpcUa/opcua-modeler #106: TypeError: hasattr(): attribute name must be string

Muddasir-Shakil, 2021-01-22:
I was creating an information model nodeset for my application and I needed some structured datatypes, so I followed the steps for creating custom structure type. But when I saved and closed my modeler then I could not reopen the .uamodel file or either import .xml file in the modeler. I get the following error:
TypeError: hasattr(): attribute name must be string
I do not have a clue how to solve that?

oroulet, 2021-01-22:
Can you share your datamodel? did you use master or pip installation?

Muddasir-Shakil, 2021-01-22:
the datamodel is here: https://github.com/Muddasir-Shakil/Opc.Ua.Ieee21451.NodeSet
I used master for installation

rekikfadwa, 2021-06-15:
I have the same problem. Did you get the solution for that?