FreeOpcUa/opcua-modeler #107: enumeration

klementuel, 2021-03-15:
Hello,
I would like to create my own enumeration type, had already tried to copy existing ones, but unfortunately I can't change the NodeId and the whole thing ends in an error.

Do you have a trick how this could be done, or is this still nice to have?

Greetings

oroulet, 2021-03-15:
using the new API from asyncua it now much easier(and safer) to create models using code. see example https://github.com/FreeOpcUa/opcua-asyncio/blob/master/examples/server-data-type-definition.py

Otherwise make a node and then copy the enum definitoin variable from another node and modify. It should work

klementuel, 2021-03-15:
ok this might be a solution for another project, in this project (this is for a thesis) I would like to use a modeler to better display the modelling. Then use the created NodeSet as a diversion for an open62541 server.

Unfortunately I don't quite understand the copy, when I copy and paste an enum type, I get the error:
`opcua.ua.uaerrors._base.UaError: Datatype must be a subtype of builtin types i=20010
`
because I can't edit the NodeID to push the type into my namespace.

If it doesn't work, that's just the way it is...

oroulet, 2021-03-15:
I have done that mamy times so it should work. Create a new data type under enums. They only copy the variable describing the values

klementuel, 2021-03-15:
I try it again, but it always only and int32 without enum informations...