FreeOpcUa/opcua-modeler #109: SyncNode' object has no attribute 'get_attributes'

MartinMajewski, 2021-05-30:
Hi there,

I have difficulty getting a proper XML model out of opcua-modeler as the ua-modelcompiler by sailavid (docker pull sailavid/ua-modelcompiler) keeps telling me, that `There is an error in XML document (2, 2).`

So I decided to look into the code and imported opcua-modeler into PyCharm Pro. I resolved all dependencies, and it is starting, but I get the following error when trying to open a new model:

```
uawidgets.utils - ERROR - 'SyncNode' object has no attribute 'get_attributes'')
Traceback (most recent call last):
  File "/Users/martinmajewski/Development/Python/opcua-modeler/venv/lib/python3.9/site-packages/uawidgets/utils.py", line 21, in wrapper
    result = func(self, *args)
  File "/Users/martinmajewski/Development/Python/opcua-modeler/uamodeler/uamodeler.py", line 204, in new
    self._model_mgr.new_model()
  File "/Users/martinmajewski/Development/Python/opcua-modeler/uamodeler/model_manager.py", line 90, in new_model
    self.modeler.tree_ui.set_root_node(self.server_mgr.nodes.root)
  File "/Users/martinmajewski/Development/Python/opcua-modeler/venv/lib/python3.9/site-packages/uawidgets/tree_widget.py", line 43, in set_root_node
    self.model.set_root_node(node)
  File "/Users/martinmajewski/Development/Python/opcua-modeler/venv/lib/python3.9/site-packages/uawidgets/tree_widget.py", line 167, in set_root_node
    desc = self._get_node_desc(node)
  File "/Users/martinmajewski/Development/Python/opcua-modeler/venv/lib/python3.9/site-packages/uawidgets/tree_widget.py", line 171, in _get_node_desc
    attrs = node.get_attributes([ua.AttributeIds.DisplayName, ua.AttributeIds.BrowseName, ua.AttributeIds.NodeId, ua.AttributeIds.NodeClass])
AttributeError: 'SyncNode' object has no attribute 'get_attributes'
```

I could not find anything meaningful on the web regarding this exact error message, but I noticed that on startup, I get the following error once:

```
QObject::connect: Cannot queue arguments of type 'QTextCursor'
(Make sure 'QTextCursor' is registered using qRegisterMetaType().)
```

Maybe my issue somehow propagates from this issue?

I searched the web and found one hint of what's possibly wrong:

https://stackoverflow.com/questions/2104779/qobject-qplaintextedit-multithreading-issues

However, this seems like an issue for some Python-Pros and nothing I can resolve on my own with the little time I have at hand.

However, I would like to know the other issue with the ua-model compiler and why it is complaining about the XML file.
Shall I open another issue for this, or is this something known that can be answered in a few sentences?

Best wishes,
Martin

oroulet, 2021-05-30:
Most probably get_attribute method was not mapped to SyncNode. Have a look at sync.py in source code and add it if missing. Maybe it is already here in master but not released yet

MartinMajewski, 2021-05-30:
I cloned the master branch so I'm definitely on the latest files there.

However, there is not sync.py file anywhere. Which one do you mean exactly?

oroulet, 2021-05-31:
Opcua-asyncio. That work is not in modeler but in a dependency

oroulet, 2021-05-31:
Opcua-modeller master requires. Opcua-asyncio master

grautvornix, 2021-07-16:
Hi,

actually I am having the same issue, so what was the resolution ? 
I found the sync.py but I am a bit unsure how to exaclty add the method, but I will give it a try and if it works will issue a pullrequest (would be my first one).

Or is it already solved and I am missing out something ?

Marc