FreeOpcUa/opcua-modeler #113: AttributeError: 'Node' object has no attribute 'read_attributes'

KalaiKumarR, 2022-01-05:
When I tried to click the "New Model" or "Open"(tried to open the xml file) in free opcua-modeler I am getting the following error called "AttributeError: 'Node' object has no attribute 'read_attributes'"

plyzico, 2022-01-07:
I've got the same error and didn't change anything. Yesterday all works fine... so would be nice, if someone can help.
Thanks

oroulet, 2022-01-07:
that sounds quite strange. nobody has released anything on that software for a while... or mabe becuae uawidget was release and broke that one...
can you try to use git master? It seems to work here.
But as written in readme it is safer and easier to create models in code

mhptk, 2022-01-07:
> that sounds quite strange. nobody has released anything on that software for a while... or mabe becuae uawidget was release and broke that one... can you try to use git master? It seems to work here. But as written in readme it is safer and easier to create models in code

i had the problem with opcua-widgets version 0.6 with downgrade to version 0.5.10 it works for me

civilizeddev, 2022-01-10:
opcua-widgets 0.6 has been released, it does not work with opcua-modeler 0.5.10.

A workaround to this is to downgrade `opcua-widgets` to `0.5.10`

If you use the poetry as a package manager, you can do this:

```
poetry add opcua-widgets@^0.5.10
```

Then it will downgrade your opcua-widgets from 0.6.0 to 0.5.10.

oroulet, 2022-01-10:
I updated the setup.py files in opcua-modeller so it does not upgrade to 0.6.0. 
But as written above, code is now a better solution

om327, 2022-02-12:
Hello, I've tried running the modeler but seem to be hiting the same issues as users above. I've checked and the opcua-widget version i'm using is 0.5.10. I've tried uninstalling and reinstalling that version but no luck - is the modeler now nolonger supprted?