FreeOpcUa/opcua-modeler #114: TypeError: __init__() got an unexpected keyword argument 'namespaceidx'

Landingit, 2022-01-13:
Hello everyone :-)

A— I try to run opcua-modeler on my ubuntu1804(Python 3.8.12)。

B— I use the method which is writen in opcua-modeler/README.md as follow:
Development 
1. git clone https://github.com/FreeOpcUa/python-opcua.git 
2. git clone https://github.com/FreeOpcUa/opcua-widgets.git
3. export PYTHONPATH=$PWD/python-opcua;$PWD/opcua-widgets  # let Python find the repositories
or set PYTHONPATH=%PYTHONPATH%;%cd%\python-opcua;%cd%\opcua-widgets on Windows
4. git clone https://github.com/FreeOpcUa/opcua-modeler.git 
5. cd opcua-modeler;
6. 'python3 app.py' # or 'make run'

C— The modeler GUI can run up, but when I try to set a new Node, the error occurs:

uawidgets.utils - ERROR - __init__() got an unexpected keyword argument 'namespaceidx'')
Traceback (most recent call last):
  File "/home/chaos/work/try/opcua-widgets/uawidgets/utils.py", line 21, in wrapper result = func(self, *args)
  File "/home/chaos/work/try/opcua-modeler/uamodeler/uamodeler.py", line 308, in add_object args, ok = NewUaObjectDialog.getArgs(self.modeler, "Add Object", self._model_mgr.server_mgr, base_node_type=self._model_mgr.server_mgr.nodes.base_object_type)
  File "/home/chaos/work/try/opcua-widgets/uawidgets/new_node_dialogs.py", line 96, in getArgs return dialog.get_args(), True
  File "/home/chaos/work/try/opcua-widgets/uawidgets/new_node_dialogs.py", line 112, in get_args nodeid, bname = self.get_nodeid_and_bname()
  File "/home/chaos/work/try/opcua-widgets/uawidgets/new_node_dialogs.py", line 82, in get_nodeid_and_bname nodeid = ua.NodeId(namespaceidx=ns)
TypeError: __init__() got an unexpected keyword argument 'namespaceidx'

D— I hava watched similar issues from：
1） FreeOpcUa /opcua-modeler.  #109 SyncNode' object has no attribute 'get_attributes' 
2） FreeOpcUa /opcua-modeler.  #113 AttributeError: 'Node' object has no attribute 'read_attributes' 
3）  FreeOpcUa /opcua-asyncio.  #633 v0.9.14 -> v0.9.90 breaks NodeId interface
Someone said the problem may be caused by the code version. I checked the version in local:
python-opcua 0.98.13;
opcua-asyncio 0.9.92;
opcua-widget 0.6.0;
opcua-modeler 0.5.12;
Then I changed the opcua-widgets to 0.5.10, and the opcua-modeler to 0.5.10 ,but it dosen't work either. 

E—— I'd like to get some suggestions here.

Best Wishes!
Mao.

swamper123, 2022-01-13:
We had a bug report on that a couple of days ago, that  ' opcua-widget 0.6.0' is causing trouble.
Downgrade it back to 0.5.10. and then try it again. Maybe it helps.

Landingit, 2022-01-13:
> We had a bug report on that a couple of days ago, that ' opcua-widget 0.6.0' is causing trouble. Downgrade it back to 0.5.10. and then try it again. Maybe it helps.

A—I have downgraded "opcua-widget" to 0.5.10. Others remains:
python-opcua 0.98.13;
opcua-asyncio 0.9.92;
opcua-modeler 0.5.12;

When I click "new model" icon on the GUI,  the follow error occurs(same as #109):
uawidgets.utils - ERROR - 'SyncNode' object has no attribute 'get_attributes'')
Traceback (most recent call last):
  File "/home/chaos/work/try/opcua-widgets/uawidgets/utils.py", line 21, in wrapper result = func(self, *args)
  File "/home/chaos/work/try/opcua-modeler/uamodeler/uamodeler.py", line 204, in new self._model_mgr.new_model()
  File "/home/chaos/work/try/opcua-modeler/uamodeler/model_manager.py", line 90, in new_model  self.modeler.tree_ui.set_root_node(self.server_mgr.nodes.root)
  File "/home/chaos/work/try/opcua-widgets/uawidgets/tree_widget.py", line 43, in set_root_node    self.model.set_root_node(node)
  File "/home/chaos/work/try/opcua-widgets/uawidgets/tree_widget.py", line 167, in set_root_node desc = self._get_node_desc(node)
  File "/home/chaos/work/try/opcua-widgets/uawidgets/tree_widget.py", line 171, in _get_node_desc attrs = node.get_attributes([ua.AttributeIds.DisplayName, ua.AttributeIds.BrowseName, ua.AttributeIds.NodeId, ua.AttributeIds.NodeClass])
AttributeError: 'SyncNode' object has no attribute 'get_attributes'

B—I have downgraded "opcua-widget" to 0.5.10, and "opcua-modeler" to 0.5.10. Others remians:
python-opcua 0.98.13;
opcua-asyncio 0.9.92;
When I click "new model" icon on the GUI,  the follow error occurs(same as #113):
uawidgets.utils - ERROR - 'Node' object has no attribute 'read_attributes'')
Traceback (most recent call last):
  File "/home/chaos/work/try/opcua-widgets/uawidgets/utils.py", line 21, in wrapper result = func(self, *args)
  File "/home/chaos/work/try/opcua-modeler/uamodeler/uamodeler.py", line 204, in new self._model_mgr.new_model()
  File "/home/chaos/work/try/opcua-modeler/uamodeler/model_manager.py", line 92, in new_model    self.modeler.tree_ui.set_root_node(self.server_mgr.nodes.root)
  File "/home/chaos/work/try/opcua-widgets/uawidgets/tree_widget.py", line 41, in set_root_node    self.model.set_root_node(node)
  File "/home/chaos/work/try/opcua-widgets/uawidgets/tree_widget.py", line 165, in set_root_node desc = self._get_node_desc(node)
  File "/home/chaos/work/try/opcua-widgets/uawidgets/tree_widget.py", line 169, in _get_node_desc attrs = node.read_attributes([ua.AttributeIds.DisplayName, ua.AttributeIds.BrowseName, ua.AttributeIds.NodeId, ua.AttributeIds.NodeClass])
AttributeError: 'Node' object has no attribute 'read_attributes'

What else can I try? 

PS：There is an old version works with Python3.5.2/win10, I have try to install that version on my ubuntu1804, but there are problems installing Python3.5.2(some important packages are not installed, pip install fails, thus I can't move on next).

oroulet, 2022-02-13:
tried to fix things today and made a relase. Let me know if it works. But anyway that software is nt maintained anymore. see https://github.com/FreeOpcUa/opcua-modeler/blob/master/README.md

oroulet, 2022-02-16:
version should be 0.5.14