FreeOpcUa/opcua-modeler #115: opcua-modeler==0.5.12 incompatible with python-opcua==0.98.13

danmoser, 2022-02-14:
On Feb 14, 2022 I've used pip to install `opcua-modeler` in Python 3.9. It picked 0.5.12 and  python-opcua==0.98.13. Then I could not work with modeler. The error `'SyncNode' object has no attribute 'get_attributes'` prevented the GUI to enable its buttons.

I've tried to install opcua-modeler==0.5.13, but then pip strangely said `ERROR: No matching distribution found for python-opcua>=0.98.13`. 

opcua-modeler==0.5.10 appears to be working well with python-opcua==0.98.13. This is my current working setup:
```
aiofiles          0.8.0
aiosqlite         0.17.0
asyncua           0.9.92
cffi              1.15.0
cryptography      36.0.1
lxml              4.7.1
opcua             0.98.13
opcua-modeler     0.5.10
opcua-widgets     0.5.10
pip               20.3.4
pkg-resources     0.0.0
pycparser         2.21
pyhdust           1.5.4
PyQt5             5.15.6
PyQt5-Qt5         5.15.2
PyQt5-sip         12.9.1
python-dateutil   2.8.2
pytz              2021.3
setuptools        44.1.1
six               1.16.0
sortedcontainers  2.4.0
typing-extensions 4.1.1
```

oroulet, 2022-02-15:
Reinstall. This is fixed a few days ago

danmoser, 2022-02-16:
No good. If I force `pip install -U opcua-modeler==0.5.13`, pip gives me the following error:
```
Collecting opcua-modeler==0.5.13
  Using cached opcua-modeler-0.5.13.tar.gz (15 kB)
  Preparing metadata (setup.py) ... done
Requirement already satisfied: opcua-widgets<0.6.0 in ./opcua-modeler-env/lib/python3.9/site-packages (from opcua-modeler==0.5.13) (0.5.10)
ERROR: Could not find a version that satisfies the requirement python-opcua>=0.98.13 (from opcua-modeler) (from versions: none)
ERROR: No matching distribution found for python-opcua>=0.98.13
```

oroulet, 2022-02-16:
sorry the last verion 0.5.14 was not uploaded for some reasons. Now it is

oroulet, 2022-02-16:
You probably forgot to update something m try to update opcua-widgeta too

danmoser, 2022-02-16:
Okay. It appears do be working now! But, I would say that version dependencies should be in the `setup.py` file...

closed 2022-02-16 by danmoser